F5OEO/rpitx #2: can't generate file from mp3 to wav to ft

besbesbechir, 2015-11-02:
I tried many times to create ft file but i get the same problem

root@raspberrypi:~/rpitx# convert -depth 8 img.jpg picture.rgb
root@raspberrypi:~/rpitx# ./pisstv picture.rgb picture.ft
Adding VIS header to audio data.
Done adding VIS header to audio data.
Adding VIS trailer to audio data.
Done adding VIS trailer to audio data.
root@raspberrypi:~/rpitx# sudo ./rpitx -m RF -i picture.ft -f 100000
Failed to read Filein picture.ft
rpitx Version 0.1 compiled Nov  2 2015 (F5OEO Evariste) running on END OF PiTx
root@raspberrypi:~/rpitx# ./pifm el.wav fm.ft
Reading file : el.wav
Sample Rate : 48000
Channels    : 1
----------------------------------------
Writing file : fm.ft
root@raspberrypi:~/rpitx# sudo ./rpitx -m RF -i fm.ft -f 100000 -l
Failed to read Filein fm.ft
rpitx Version 0.1 compiled Nov  2 2015 (F5OEO Evariste) running on END OF PiTx

F5OEO, 2015-11-02:
Maybe some right issue : you seems to be root. Can you test with normal user ?

besbesbechir, 2015-12-05:
im always getting failed on reading file 

pi@raspberrypi ~/rpitx $ convert -depth 8 picture.jpg picture.rgb
pi@raspberrypi ~/rpitx $ ./pisstv picture.rgb picture.ft
Adding VIS header to audio data.
Done adding VIS header to audio data.
Adding VIS trailer to audio data.
Done adding VIS trailer to audio data.
pi@raspberrypi ~/rpitx $ sudo ./rpitx -m RF -i picture.ft -f 100000
Failed to read Filein picture.ft
rpitx Version 0.1 compiled Nov  2 2015 (F5OEO Evariste) running on END OF PiTx
pi@raspberrypi ~/rpitx $

F5OEO, 2015-12-14:
Can you check that picture.ft exists and is not zero size ?

F5OEO, 2016-09-01:
Any answer ? Else will be closed

closed 2016-09-02 by F5OEO