F5OEO/rpitx #3: cover dcf77

gaionim, 2015-11-05:
Hi,
it is possible extend frequency to 77 Khz ?
If yes,
 I like to try  something like https://github.com/CodingGhost/DCF77-Transmitter/blob/master/DCF77_Protocoll_.ino
With some hint I can do the job.

F5OEO, 2015-11-07:
yes, it seems to be possible..I have to modify the soft..I let you know when it is available

gaionim, 2015-11-08:
Il 07/11/2015 12:17, F5OEO ha scritto:
> yes, it seems to be possible..I have to modify the soft..

thanks, I assume it will possible soon or later.

For more general lower frequency see
https://en.wikipedia.org/wiki/Radio_clock

If no problem with hw/sw
60  kHz will make happy U.S. people
and some russian and cina too.
25 kHz for everybody ;-)

Do you have some hint for me to modulate from 100% to 15% ?

In the next days I'll watch the code to see if I'm capable to add dcf77
features.

Regards, Mauro

F5OEO, 2015-12-14:
rpitx is now down to 4KHZ..Up to write the code to make DCF77 !

gaionim, 2015-12-14:
Thanks, hope to have spare time in xmas.
See you next year ;-)

gaionim, 2015-12-15:
Hi,
can I ask what is the best way to
"..reducing the amplitude of the carrier to 15% of normal (−16½ dB) for 0.1 or 0.2 seconds...." ?

I think to write a RFA file , but which value of frequency and amplitude I have to use
for "normal amplitude carrier" and which for " 15% reduced amplitude" ?

The transmitting frequency will be 77.5 kHz.

F5OEO, 2015-12-15:
Amplitude is really not precise for the moment and dynamic is very limited. Hopefully DCF77 is not really sensitive to amplitude but mostly to time of pulse. So using RFA is the easiest way : send Amplitude and Time. When you modulate it with rpitx you will set the frequency.
I can maybe try push a draft version if you like but not tested as I don't have a receiver.

gaionim, 2015-12-15:
so, what is the value to put in amplitude ?
I agree not critical , but need a value for 100% and one for 15%
Doc says
"{RFA(FileInput is a (double)Frequency,(int)Time in nanoseconds,(float)Amplitude}"
I have to write 
8bit(double) + 4 bit ( int) + 4bit ( float)  ?

I can read the code, but my C is old and "rusty" ;-)

I think to write the file in python:
 -  read the datetime
 - convert in 59 bit format the next 10 minutes (or so) with shift of one minute
 -  write the file with the data, 
 - then whait start of minute and call  rpitx

I have a receiver clock to test , the same that take some day to fix  summer daylight-saving
and a Pi to test ( only need to find it )

F5OEO, 2015-12-15:
I just commit a blind implentation (not tested at all). Time is fixed to 7min59, just a proof of concept to debug !

gaionim, 2015-12-15:
thanks, this weekend I'll look for my pi to test and I'll report.

gaionim, 2015-12-15:
just looking the code, but seems uncorrect,
maybe something like

void modulate(byte b)
{
    if (b == 0)
   {
     playtone(32767/4,100000000);
     // playtone(32767,900000000);  // not sure needed
   }
   else
   {
	playtone(32767/4,200000000);
	// playtone(32767,800000000);  //  not sure needed
   }
}

see https://en.wikipedia.org/wiki/File:DCF77_code.svg

F5OEO, 2015-12-15:
yes right, I didn't well read the original code and too lazy too read the documentation. I modify it

gaionim, 2015-12-16:
Forked , see https://github.com/gaionim/rpitx/tree/dcf77/contrib
NOT TESTED yet, 
hope this weekend test and fix on my Pi

gaionim, 2015-12-19:
Test dcf77 failed :-(
fm  ( at 78 MHz)  and ssb ( at 101 kHz )  are ok, but dcf77 is ko

$ ./pidcf77 /tmp/upstream.bin 
$ sudo rpitx -m RFA -i /tmp/upstream.bin -f 77.5 -l

clock do not see nothing,   s-meter  is 0 all the time

Some note,
Try to see signal with DSO ( http://www.seeedstudio.com/wiki/DSO_Nano ) , but not so skilled.
Connected probe to wire with 4500 nF capacitor to filter 50 Hz component.
I have problem to lock signal, so frequency is changing, but max seems correct, around 77-78 kHz 

Amplitude is around .4 mV for  32767   and 0 for 0 ,
but is the same with  10000 and 1500 ( 100% and 15% from specs).
I do not know if is my bad skills with DSO or is real amplitude ( on or off ?!? )

I'll try with different frequency and longer wire, but not optimist :-(

F5OEO, 2015-12-19:
I make some time correction in the last commit, but don't think it is the
issue. Guess that frequency and timing that is used with rpitx does not
suit well for Very Low Frequency. I have to make a special mode for VLF
(under 1 MHZ)

gaionim, 2015-12-19:
thanks, I'll fix binary encoding and wait for VLF mode.

NTMMFTS, 2016-05-28:
hi, any progress on VLF mode?  I'd probably be the first person to sync a watch in Hawaii if this works :-)

gaionim, 2016-09-03:
@NTMMFTS 
try now, my clock do not work ( see https://github.com/F5OEO/rpitx/pull/36 )
but seems F5OEO's does.
Let me know if work for you,
simply update and
pi@raspberrypi:~/rpitx $ python setTime.py

gaionim, 2016-09-04:
@NTMMFTS 
now works for me too.

closed 2018-11-23 by F5OEO