F5OEO/rpitx #5: Rpi A+

patriciogonzalezvivo, 2015-11-15:
Hi, thanks for sharing this great project!
I discover that it officially doesn't have support for the A+ model. Why is that? because of memory? do you think there is going to be A+ version working on the future?

Thanks again and congratulations for the good work.

Patricio

F5OEO, 2015-11-22:
A+ should work, but not have a unit to test

patriciogonzalezvivo, 2015-11-23:
Hi! I tested and it doesn't seams to work. The program finish before sending anything. I will test again giving more memory to the CPU.

patriciogonzalezvivo, 2015-11-24:
This is what I get on my A+

```bash
$ ./snapsstv.sh                                                                                                                                                                                                   *[master]  
Adding VIS header to audio data.
Done adding VIS header to audio data.
Adding VIS trailer to audio data.
Done adding VIS trailer to audio data.
rpitx Version 0.1 compiled Nov 14 2015 (F5OEO Evariste) running on  Model Pi 2
Resolution(Hz)=18.366667
END OF PiTx
```

 
[image]

ha7ilm, 2015-11-28:
Hi!
It seems to me that `piBoardId` identifies this board as Pi 2 erroneously.
Could you post the output of `cat /proc/cpuinfo` on the A+?

ha7ilm, 2015-11-28:
You could also try this temporary modification whether it fixes your problem:

Replace this line over here:

https://github.com/F5OEO/rpitx/blob/master/src/RpiGpio.c#L17

...with the following:

    if(false)

patriciogonzalezvivo, 2015-11-30:
Hi! Thanks for the answer and temporal fix.
Here is the ```/proc/cpuinfo```

```bash
[~]$ cat /proc/cpuinfo                                                                                                                 
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 2.00
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2708
Revision	: 0012
Serial		: 000000009ae35ba7
```

ha7ilm, 2015-12-01:
Thanks. I'm going to make a test with it as soon as I have time.

Did the temporary fix work?

patriciogonzalezvivo, 2015-12-01:
Yes! it compiles and sends something that I can catch with my rtl_sdr. ( I'm sending images trough sstv script - great work btw!! Still need to learn how to capture it back : ) )

Thanks!

tomswartz07, 2015-12-02:
@patriciogonzalezvivo to capture it back, you'd be well suited to use a RTL-SDR; which is a modified USB tv tuner that can capture a wide range of radio frequencies. You can find them for about $12 on Amazon.
:radio: 

My personal setup has [this SDR](http://www.amazon.com/RTL2832U-Low-Cost-Software-Compatible-Packages/dp/B00SXZDUAQ/ref=sr_1_4?ie=UTF8&qid=1449060125&sr=8-4&keywords=rtl+sdr) and I use the program GQRX on my laptop to view the radio freqencies.

Hope this helps!

F5OEO, 2016-08-27:
Model detection should be fixed : https://github.com/F5OEO/rpitx/commit/feb6f4cf3e62f013688671abd5d365b6753baf1e

closed 2018-11-23 by F5OEO