F5OEO/rpitx #6: Does not seem to Tx

haxorthematrix, 2015-12-10:
Raspberry pi B+ with wire attached to appropriate gpio pin. Receiving with RTL-SDR and HDSDR under windows, tuned to 107Mhz (an unused FM station,  low, low power FM Tx is legal in the US).

I cannot observe any Tx from the pi with the RTL-SDR, using either:

sox imperial_march.wav imp.wav rate 48000 channels 1
pifm imp.wav imp.ft
sudo rpitx -m RF -i imp.ft -f 107000

or 

convert -depth 8 Picture.jpg picture.rgb
./pisstv picture.rgb picture.ft 
sudo ./rpitx -m RF -i picture.ft -f 107000

Both commands return, almost immediately (less than 1 second delay) with:

rpitx Version 0.1 compiled Dec  9 2015 (F5OEO Evariste) running on  Model Pi B+
Model Pi 1
END OF PiTx

I have attempted with and without the patch from Issue #1.
The pifm project, on the same pi works perfectly with the antenna attached to a different GPIO pin, as required by the pifm project.

F5OEO, 2015-12-14:
Try latest version...it detects if you are running Wheezy or Jessie which seem to be the problem (DMA channel)

haxorthematrix, 2015-12-16:
SSTV appeared to transmit, but was only able to verify with the HackRF One and Portapack waterfall display.  I will investigate further.

redfast00, 2015-12-17:
Same here: pifm works great, but rpitx doesn't seem to transmit (tested with rtl-sdr dongle)

F5OEO, 2015-12-17:
When there is an issue : please report console output and details of issue. Just saying doesn't work is not helpfull to fix it !

haxorthematrix, 2015-12-17:
Ok, more investigation.  With the latest pull, transmission appears to be better, specifically with SSTV:

sudo ./rpitx -m RF -i picture.ft -f 103000
rpitx Version 0.1 compiled Dec 16 2015 (F5OEO Evariste) running on Master PLL = 
1000000000
 Model Pi B+
Model Pi 1
Jessie
MASH 1
SampleRate=100000
WaitNano=25000 F1=205994769.664052 TuneFrequency 206000000.000000 F2=206005130.0
10562 Initial Resolution(Hz)=10360.346510 ResolutionPWMF 65.159412 NbStep=159 DE
LAYStep=14
****** STARTING TRANSMIT ********
END OF PiTx

Tx takes about 4-5 seconds.  I can verify that signal is recieved by an RTL-SDR, as the SDR# waterfall display changes, but fails to decode with MMSSTV (the decode may be my problem however, I am still experimenting)

redfast00, 2015-12-17:
It says it's transmitting, but I don't see anything in gqrx. When I use pifm (from pifm project), I do see and hear an fm-signal.

redfast00, 2015-12-17:
Is this related to the difference in pin used? (pin 12 vs pin 7)?

ha7dcd, 2015-12-19:
Of course! Its a completely different signal source, you need to use the
propet gpio for rpitx and for pifm as they use different pins

redfast00, 2015-12-19:
When using pifm, I don't attach an antenna and still receive it fine

F5OEO, 2015-12-19:
Can you give what the output of your command and the result rpitx give ...

haxorthematrix, 2016-09-01:
Ok, coming back around to this one.  I don't know what happened, but I have it working.  Changed SSTV decoder to QSSTV and it worked great.  I did so many things in the mean time, including a git pull recently that who knows.  It works now!

OutsourcedGuru, 2017-05-18:
Like @redfast00 suggested, within several feet of your project the standard Raspi pins are long enough to transmit without any antenna whatsoever.

I've been known to put my oscilloscope probe into one end of my antenna and then "walk" the far end of that [antenna wire](http://img.dxcdn.com/productimages/sku_55454_4.jpg) from pin to pin until it shows me the correct output pin (12, in my case).

closed 2018-11-23 by F5OEO