F5OEO/rpitx #14: Raspberry Pi 3 support

xmikos, 2016-03-01:
This is more of a question than an issue - do you think `rpitx` should work without modifications on new Raspberry Pi 3? Or if not, are you planning to add support for it?

CPU in RPi3 runs on higher frequency, so maybe rpitx should work better for transmitting on higher frequencies too?

F5OEO, 2016-03-01:
I wait for my pi 3 to test. So I cant' tell if rpitx works or not. 
About higher frequency, I have maybe some ticks to get higher even on Raspberry 2 but not yet tested.

jnk0le, 2016-03-01:
Broadcom have to release BCM2837 datasheet first, if there was a change like pi1 -> pi2.

freebil, 2016-04-04:
Do you have any news about raspberry pi 3?

primeshp, 2016-04-10:
I have verified its working with Pi 3. Any news of this working on Pi Zero. I could not make it work

E3V3A, 2016-04-17:
@primeshp  Any ideas/tests regarding the upper freq limit for PWM? (The old rp2 has about 125 MHz according to BCM2835 datasheet.)

OutsourcedGuru, 2017-05-18:
Works great on my Raspberry Pi 3 Model B, pin 12, Raspbian Jessie (pretty sure it's the Lite version without the GUI).  I don't think I had to `sudo raspi-config` to change the **Interfacing Options** for this to work.  Pretty sure I did a `sudo apt-get update`, however, even after the install script for rpitx (which I think includes one at the top).

I managed to get it to stream realtime from a USB-based microphone, noting that my [Radio Shack digital mic](https://www.radioshack.com/products/digital-usb-clip-on-microphone?gclid=CPXv94-k-tMCFRBEfgodklQJ6A) shows up as **plughw:1**, transmitting mono on FM 87.9 Mhz:

```
arecord -D plughw:1 -f S16_LE -r 48000 /dev/stdout 2> /dev/null | ./pifm /dev/stdin /dev/stdout | sudo ./rpitx -m RF -i /dev/stdin -f 87900
```

OutsourcedGuru, 2017-05-18:
And btw, this project next will be ported over to a **Raspberry Pi Zero W** after the initial shake-down test so I should know more after testing this at the venue first to their 548.5 Mz—tuned [Sennheiser](https://static.bhphoto.com/images/images500x500/Sennheiser_XSW_35_B_XSW_35_Vocal_Set_1328552661000_842146.jpg) receiver.

abiank, 2017-06-04:
On my unmodified Rpi3 with the latest jessie full the "main" output frequency is a bit off. For example:

sudo ./rpitx -m RF -i fm.ft -f 429000 -l
rpitx Version 0.2 compiled Jun  4 2017 (F5OEO Evariste) running on Master PLL = 1000000000

 Warning : Using harmonic 3
memory: 1024 MB
processor: Broadcom BCM2837
i2cDevice: /dev/i2c-1
model: Model B Pi 3
manufacturer: Sony
pcb revision: 2
warranty void: no
revision: a02082
peripheral base: 0x3f000000
Jessie
Using mbox device /dev/vcio.
3616000 Size NUM PAGES 883 PAGE_SIZE 4096
MASH 1 Freq PLL# 6
Calibrate : ppm=-7 DMA 175ns:1592ns
WaitNano=15908 F1=95331192.105386 TuneFrequency 95333298.665365 F2=95333410.915862 Initial Resolution(Hz)=2218.810476 ResolutionPWMF 24.653450 NbStep=90 DELAYStep=9
****** STARTING TRANSMIT ********

gives the following output as received by gqrx:

[image]

I ask for 429.000Mhz but I get the signal on 430.665500Mhz . The only thing that's remarkable about this pi3 is that it is connected to a non hdmi monitor; should i try another one?

Sceleratis, 2018-09-13:
> 
> 
> On my unmodified Rpi3 with the latest jessie full the "main" output frequency is a bit off. For example using the provided sample wav file:
> 
> sudo ./rpitx -m RF -i fm.ft -f 429000 -l
> rpitx Version 0.2 compiled Jun 4 2017 (F5OEO Evariste) running on Master PLL = 1000000000
> 
> Warning : Using harmonic 3
> memory: 1024 MB
> processor: Broadcom BCM2837
> i2cDevice: /dev/i2c-1
> model: Model B Pi 3
> manufacturer: Sony
> pcb revision: 2
> warranty void: no
> revision: a02082
> peripheral base: 0x3f000000
> Jessie
> Using mbox device /dev/vcio.
> 3616000 Size NUM PAGES 883 PAGE_SIZE 4096
> MASH 1 Freq PLL# 6
> Calibrate : ppm=-7 DMA 175ns:1592ns
> WaitNano=15908 F1=95331192.105386 TuneFrequency 95333298.665365 F2=95333410.915862 Initial Resolution(Hz)=2218.810476 ResolutionPWMF 24.653450 NbStep=90 DELAYStep=9
> ****** STARTING TRANSMIT ********
> 
> gives the following output as received by gqrx:
> 
> [image]
> 
> I ask for 429.000Mhz but I get the signal on 430.665500Mhz . The only thing that's remarkable about this pi3 is that it is connected to a non hdmi monitor; should i try another one?
> 
> I tried sudo -s echo powersave > /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor as suggested in issue #21 but it didn't help. Let me know if i can run any needed test.

Having this issue as well currently and it lead me here. Dunno what to do. Trying to replay a recorded signal on 433890400 but it's transmitting it at extremely poor quality around 434000000 instead. Also GPIO pin 12 doesn't seem to work at all with rpitx so I've had to use -c 1 to get any results at all. Pi_FM_adv has no issue transmitting an fm signal at the target frequency so I'm pretty sure it's not the fault of the pi itself. Running this on a pi3.

closed 2018-11-23 by F5OEO