F5OEO/rpitx #15: Code needs cleanup.

SopaXorzTaker, 2016-04-02:
I understand that this project is only an experiment, but its code is barely readable.

F5OEO, 2016-04-02:
Patch is welcome !

closed 2016-04-02 by F5OEO

csete, 2016-04-02:
@SopaXorzTaker Not sure why you think the code is barely readable. If it is because of the formatting, there are plenty of freely available tools that can format the code according to your preferred style, e.g. indent, uncrustify and clang-format just to name a few.

SopaXorzTaker, 2016-04-04:
@csete, I guess that's my fault then. I just have different C code style expectations.

SopaXorzTaker, 2016-04-04:
Probably worth to run a reformatting tool.
I was talking about some styling issues: e.g. https://github.com/F5OEO/rpitx/blob/master/fm/pifm.c

natevw, 2016-04-20:
Almost looks like the main thing here might be simply mixed tabs vs. spaces? That might be nice to get cleaned up via a PR as maintainer says.

SopaXorzTaker, 2016-04-21:
Not only, there also are issues like "int a=5;" (no spaces around), and
there is code that seems dead.