F5OEO/rpitx #16: RX Problem

virtualrobotix, 2016-04-08:
I try to decode the stream using standard DVB-S receiver ... but when i search the channel the receiver don't found my modulation ... I try to check the modulation with HackRF and #SDR amd i see the modulation but i don't have any kind of software for decode DVB-S with HackRF i search online but found nothing ... do you have some suggestions ? If not i order a RTL-SDR ... but i thought that with HackRF i don't need a RTL dongle help me please :) 
Thanks 
Roberto

F5OEO, 2016-04-08:
Try leandvb which input a pipe in IQ from rtl or HackRF (check if it is int or float32).

virtualrobotix, 2016-04-08:
Thanks F5OEO do you have some example how i can use HackRF in input as leandvb IQ format to leandvb pipe ? I'm never used it in that way until now.
Thank

virtualrobotix, 2016-04-08:
But it can work in realtime or only in log data ?

SopaXorzTaker, 2016-04-19:
It is realtime.

F5OEO, 2016-09-02:
Closed : relative to rpidatv, not rpitx

closed 2016-09-02 by F5OEO