F5OEO/rpitx #18: Readme: Is this a typo?

E3V3A, 2016-04-17:
In the Readme.md usage you say: 

> -f float      frequency to output on GPIO_18 pin 12 in khz : (130 kHz to **750 MHz**),

Is [this](https://github.com/F5OEO/rpitx/blob/master/src/RpiTx.c#L979) a typo or can you actually have that high a frequency? In the BCM2835 manual it is stated that the PWM can be most 125 MHz, given the old document assumption on clock freq etc.

Have anyone actually tested this? 
How does it differ for different devices?

F5OEO, 2016-04-17:
PWM and PCM could be up to 500MHZ using the right PLL.
With PWM there is a divider by 2 to make a waveform : thus 250MHZ.
When frequency is above 250MHZ, rpitx compute the harmonic needed : for example -f 300000 will output 100MHZ but you could listen to the signal on 100*3=300MHZ

F5OEO, 2016-08-27:
After https://github.com/F5OEO/rpitx/commit/feb6f4cf3e62f013688671abd5d365b6753baf1e frequency fundamental is now up to 500MHZ

E3V3A, 2016-09-07:
Alright. Thanks.

closed 2016-09-07 by E3V3A