F5OEO/rpitx #21: At RPi 3 rpitx randomly shifting frequency during TX

ha7dcd, 2016-04-22:
I've experienced a strange shifting of 2.8 MHz upwards from the carrier. Randomly the nominal 14MHz transmission is just shifted up to 16.8 MHz... then back.

 Using latest Raspbian Jessie (as of today 22/04/2016), all the update, upgrade, disti upgrade done before. Cloned and installed riptx just right now from github.

[image]
[image]

freebil, 2016-04-22:
Hello! How did you see these diagrams? Thanks.

ha7dcd, 2016-04-22:
Using this: https://signalhound.com/products/usb-sa44b/
However you can catch the signals also with RTL-SDR or SDRPlay(better) too..

freebil, 2016-04-22:
Ok, thank you!

tomswartz07, 2016-04-22:
Perhaps I'm reading the graphic incorrectly, but both images show Mkr1 and Mkr2 having the same frequency.

It appears that the only difference between the two images is the center frequency.
Image 1 has a center freq of 25MHz and Image 2 has a center freq of 14MHz.

Am I correct in reading this?

ha7dcd, 2016-04-22:
No, you are mistaken the fact that the Mkr1 is the intended transmission (14 MHz) Mkr2 showing the randomly shifter carrier (16.8 MHz)

The reason for two screenshot is the different span

ha7dcd, 2016-04-22:
Maybe this video gives a better understanding of this phenomena:
[https://www.dropbox.com/s/xxihgxrakkejlji/14MHz_and_random_16M8_shifting_RPi3_gpio18.png.MP4?dl=0](url)

ha7dcd, 2016-04-24:
My friend András, @ha7ilm noticed that this shift issue possibly caused by the dynamic CPU frequency change algorithm in RPi 3. I've tested it and looks like a workaround for the problem:
" sudo -s
echo powersave > /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor "

[image]

ha7ilm, 2016-04-24:
It seems like this is caused by the `cpufreq` kernel module, which changes the frequency in order to save power. The `ondemand`governor is enabled by default.

F5OEO, 2016-07-01:
force_turbo=1 in /boot/config.txt shoud solve the issue

jnk0le, 2016-07-01:
The problem is frequency of PLLC (used by rpitx), which drifts with overclock.

F5OEO, 2016-08-27:
Should be fixed with https://github.com/F5OEO/rpitx/commit/feb6f4cf3e62f013688671abd5d365b6753baf1e : PLL_D is used which is very stable compare to PLL_C (variable)

closed 2018-11-23 by F5OEO