F5OEO/rpitx #22: Square wave / harmonics

inspector71, 2016-05-22:
I've practically no idea how radio transmission works on the Pi but I would like to use it to push music to my single-CD, AM/FM car audio unit that lacks an AUX input.

However the issues with interference concern me quite a lot. 

Is there a way to minimize the interference in software, through a super-short antenna wire, or perhaps with a super-cheap piece of hardware? 

This board seems aimed at addressing the interference issue:

https://www.tapr.org/kits_20M-wspr-pi.html

but it's more expensive than a Pi when shipping is factored in.

May sound simplistic but would a steel case, kinda acting as a faraday cage, around the Pi, be enough?

I really only need to transmit maybe 30CM max which is more than the 10CM available without adding a GPIO wire but much less than the 100M apparently attainable with the shortest of GPIO wires.

jnk0le, 2016-05-22:
You just need a simple low pass filter to get rid of unwanted frequencies.

SopaXorzTaker, 2016-05-22:
IMHO, currently PiFmRds works better, not RpiTx.

ha7dcd, 2016-05-23:
Usually a simple low pass filter doesn't helps as the whole spectrum rich
in spuriouses when you take a look at closer look from spectrum analyzer.
You better build up your band pass filter for it and analyze before going
online.

You can find here more information about the 20m shield design:
https://www.tapr.org/pdf/DCC2015-QRPi-HA7DCD.pdf

inspector71, 2016-05-23:
The task becomes too expensive if a filter is needed. I'm better off gutting the existing OEM unit and replacing it with a Pi plus a decent amplifier HAT. That way the money I'd waste on a filter gets spent on a useful HAT.

All the best.

closed 2016-05-23 by inspector71