F5OEO/rpitx #24: RPI 3 ...transmission start and stop, then restart..fake colors on sstv

michelinok, 2016-07-01:
Hi,
I own only a raspberry 3 and i've noticed these problem (i think it's something on my setup since it works for you...)

- The transmission of sstv start and stop randomly during a transmission
- The colors are not the same of the original image (tryed the raspberry cam and an user provided image).
- Does it transmit as Martin 2 mode?

Any idea?

F5OEO, 2016-07-01:
With Rpi3 there is CPU save mode which break intermitently the transmission. The easiest way is to set  force_turbo=1 in /boot/config.txt
About SSTV, I just put it as a test and other mode could be easily implemented. As I remember the mode is Martin 1.

michelinok, 2016-07-01:
Quite strange...enabling turbo it doesn't transmit anything, reverting to non-turbo it transmit (but of course same problem as before)
Any idea?

F5OEO, 2016-07-01:
disable_pvt=1
force_turbo=1
arm_freq=700
gpu_freq=250
core_freq=250
sdram_freq=400

Is my configuration ... Have to check really on Rpi3 !

jnk0le, 2016-07-01:
The problem is frequency of PLLC (used by rpitx), which drifts with overclock.

michelinok, 2016-07-02:
F5OEO, that's exactly my config :(

michelinok, 2016-07-10:
Ok,,,just tryed on raspberry 1....it doesn't start/stop randomly, but images are totally wrong, cannot see anything related to the original image (tryed  with raspicam too...)
Can someone send me a .ts image to test? maybe that's the problem...

jnk0le, 2016-07-11:
Have you tried other modes? How does 100MHz FM carrier looks like ?

michelinok, 2016-07-30:
Sorry for the big delay...my rx was k.o....
Other modes are ok, no way to see the 100MHz because i can do only 0-30Mhz :°(
Any idea? I really really would like to enjoy sstv.

michelinok, 2016-08-01:
any idea?

F5OEO, 2016-08-27:
Shoud be fixed with https://github.com/F5OEO/rpitx/commit/feb6f4cf3e62f013688671abd5d365b6753baf1e

michelinok, 2016-08-27:
> Shoud be fixed with feb6f4c

Going to try sooooooon (does it works with rpi 1?)

F5OEO, 2016-08-27:
Not tested with rpi1 (all mines burnt) but theorical work (quiet same as Pizero which work)

closed 2018-11-23 by F5OEO