F5OEO/rpitx #29: Glitchy output on RaspPi 3

natevw, 2016-08-10:
When I do `sudo ./rpitx -m VFO -f 1000` is it right to expect basically a square wave at 1000 kHz?

Instead, I get really sporadic pulses, as shown in [this video](https://twitter.com/natevw/status/763233882841817088).

Is this some sort of Pulse-Density Modulation trick or similar, or is something wrong? Mostly just trying to wrap my head around this. I am able to hear the FM sample via a broadcast receiver but when I lower the frequencies to what my 100MHz scope is supposed to be able to pick up, I don't see what I would expect.

natevw, 2016-08-11:
After trying just the `disable_pvt=1` by itself, I tried adding all the settings recommended in https://github.com/F5OEO/rpitx/issues/24#issuecomment-230030147. Still pretty much the same behavior, maybe a tiny bit better but could be my imagination.

Interestingly, after changing the settings when I run `vcgencmd get_config disable_pvt` it says "disable_pvt is unknown"! The other settings from #24 discussion are shown, e.g. it shows me that "force_turbo=1" and the freq settings as expected. But *not* `disable_pvt`.

I found https://github.com/raspberrypi/firmware/commit/ac0c0bff7951a39d371a46c4a80a9ec560e33c16 which commit message says this option might be removed?

natevw, 2016-08-11:
The VFO mode actual seems to produce a nice square wave for frequencies between about 2.5 kHz and 500 kHz. Somewhere above there, the glitches start getting worse and worse.

I wouldn't rule it completely out, but I don't think this is just a triggering/scale artifact on my oscilloscope. I can leave the horizontal scale the same between 500 kHz and 800 kHz. At 500 kHz it's not "rock solid" but it's pretty good. At 800 kHz the glitching is very apparent.

I'll share a video via Twitter again and update this when it's posted.

natevw, 2016-08-11:
I tried my card in a Raspberry Pi 1 — same things! (I did do `make clean` in src and rebuild just in case, but didn't help either.)

So I ran git bisect.

[image]

After https://github.com/F5OEO/rpitx/commit/f34233268f384aaec8005196c4bffa501a7ea637 was applied, I start seeing the "glitches" at e.g. 27125 kHz VFO output, and FM transmit at 103300 gets very very distorted. If I checkout https://github.com/F5OEO/rpitx/commit/a274a8f18969248f417f0224b540874215caf8d0 immediately before the "Add GPCLK output" commit, then the output FM example at 103300 it is actually fairly clean in the local spectrum!

natevw, 2016-08-12:
As suggested [via Twitter](https://twitter.com/F5OEOEvariste/status/763633530366812160), I tried sending the signal to GPIO 04 (pin 7) by adding the `-c 1` option on the latest master currently ca94b38c1ad72a5d281dd26c0a0e3d8cf75588bd. This actually looks really good, even compared to  a274a8f!

[image]

It's certainly not perfectly clean on the RTL-SDR "spectrogram" but looks the best I've seen it yet, from 5 kHz on up, when viewed on the oscilloscope. I'm not sure exactly what this option does (okay — I still am quite fuzzy on how the original PE1NNZ prototype even works :-) but it seems to be the right one right now!

freebil, 2016-08-22:
Hello. I tried sudo ./rpitx -m VFO -f 1000 and I take nice square waves without glitches

[image]

F5OEO, 2016-08-27:
@freebil raspberry model , Linux version ?

freebil, 2016-08-27:
@F5OEO Raspberry pi 3 with the latest raspbian jessie lite.

closed 2018-11-23 by F5OEO