F5OEO/rpitx #31: IQFLOAT isn't parsed corretly or not documented enough

spaivaras, 2016-08-23:
Hello, I just can't get rpitx to work with IQFLOAT data (complex type of GNURadio) to work properly with rpitx. I have a simple FM transmitter under gnu radio and pipe complex floats via network with sampling rate of 190000 (doesn't really matter as no sampling rate is working)

If i use
```
 while true; do (nc -l -p 8011; dd if=/dev/zero bs=4096 count=30); done | sudo rpitx -i- -m IQFLOAT -f 99000 -c 1 -s 190000
```
and i get an to wide output with 2 peaks in them, the signal is unusable:
[image]

If I try the exact same IQ data, but first passing it via csdr
```
while true; do (nc -l -p 8011; dd if=/dev/zero bs=4096 count=30); done | csdr fmdemod_quadri_cf|csdr gain_ff 75000 |csdr convert_f_samplerf 5263 | sudo rpitx -i- -m RF -f 99000
```

I get a much better signal (actually quite good one) proper width:
[image]

It would be enough to use csdr, but it introduces additional latencies, adds requirement for gain multiplication and in general would be nice if rpitx would properly handle IQ data itself :)

In README.md about streaming from GNURadio its written to "shift" the signal by 5khz, and thats centers the signal on one of 2 peaks. But that doesn't fix the issue that signal itself is not a good representation of passed IQ values

spaivaras, 2016-08-23:
Specs:
Raspberry Pi 2 B
Distro Minibian (jessie)
compiled from master

F5OEO, 2016-08-24:
The IQFLOAT type is float : be sure to not have a double float output from gnuradio.
As IQ is convert to phase/frequency, value around zero are subject to round errors. That's why you  should shift the signal. 
Agree, IQFloat is not well documented : TODO !

ha7ilm, 2016-08-24:
IQFLOAT means 32 bit interleaved floats, like:

    I(4 bytes) Q(4bytes) I(4 bytes) Q(4bytes) I(4 bytes) Q(4bytes) I(4 bytes) Q(4bytes) ...

Endianness depends on the system, it should be LE on both an RPi and a PC.

Also the signal should have the maximum amplitude possible (I mean that you should set the gain correctly in the appropriate GNU Radio blocks), but it should never exceed the [-1.0, -1.0] range. This is the line in the code where the data is converted to 16 bit signed imtegers:

https://github.com/F5OEO/rpitx/blob/master/src/RpiTx.c#L1308

However, if you want to generate an FSK or an FM signal, I suggest that you should always use `rpitx` in RF mode. Then the PWM will not be used and the signal will be cleaner. 

@F5OEO Evariste, are these correct?

F5OEO, 2016-08-24:
Yes , thx Andras for these nices explanations. Have some good news : works
on pizero and now surely on rpi3 ! Could go also upper in frequency .

ha7ilm, 2016-08-24:
> works on pizero and now surely on rpi3 ! Could go also upper in frequency .

Great!

spaivaras, 2016-08-24:
Thank you for the explanation, its exactly what I thought and what GNURadio complex type actually is, 2 floats each 32bits (no doubles). I am probably missing a single gear somewhere in my approach :) as it seems no one else had issues with IQFLOAT. I will poke more around with simple sine waves to get a better idea what actually happens

F5OEO, 2016-09-02:
@spaivaras Do you succeed in using IQFLOAT with gnuradio ?

spaivaras, 2016-09-02:
@F5OEO Nop, Tried everything, pass data thru network, read it directly from file. Lower sampling rates, higher sample rates, gains, swapping I and Q data, shifting from center. The signal always seems very distorted (like 2 carrier frequencies). Checked the data that is coming thru, everything seems fine with setup. RF modes works quite well, just can't generate a RDS signal with RF mode. I can even use IQ demodulation in gnu radio itself and pass resulting floats via csdr with convert_f_samplerf and a constant for timing. And it works. But no success with IQFLOAT. In any case I think this can be closed, as no one else seems to have any issues (or just didn't try it). I will sniff in code at some point, if I find something I will create a PR :)

F5OEO, 2016-09-04:
Have to check this IQLOAT issue coming from external tool like gnuradio. Can you send me an example of file with IQFLOAT which is distorded to inspect and reproduce here ?

spaivaras, 2016-09-04:
Sure, I generated 3 files, which can be played with no problem in GQRX or other sdr software 
https://www.dropbox.com/sh/37bybfdsh2xa688/AACyNhk-O1mO6X_7rdfp-7zba?dl=0
File names say, what settings they have

closed 2018-11-23 by F5OEO