F5OEO/rpitx #35: pidfc77

michelinok, 2016-08-28:
Hi, what kind of file is dcfpatern.bin required by pidcf77 ?

michelinok, 2016-08-28:
sorry...i meant... how to use pidcf77?

F5OEO, 2016-09-02:
pidcf77 result.rfa
then 
./rpitx -m RFA -f 77.5 -i result.rfa -c 1
But maybe an issue with the generator. Use python script instead.

michelinok, 2016-09-02:
I'll try to use the python as you suggest, but since i'm not a good developer, i cannot understand on what datetime it wanna set my clock.
On future release, it would be great if we can pass date/time as an argument to the script (it's only a suggestion!!!).
Great work! many thanks!

michelinok, 2016-09-02:
tryed the script but i get these errors:

./setTime.py: line 11:
From https://github.com/gaionim
inspired by https://github.com/CodingGhost/DCF77-Transmitter

reference
https://en.wikipedia.org/wiki/DCF77

: No such file or directory
import: unable to open X server `' @ error/import.c/ImportImageCommand/364.
import: unable to open X server `' @ error/import.c/ImportImageCommand/364.
from: can't read /var/mail/subprocess
from: can't read /var/mail/ctypes
./setTime.py: line 18: syntax error near unexpected token `('
./setTime.py: line 18: `class Sample(Structure):'

i'm not running X...i'm headless :)

F5OEO, 2016-09-02:
python ./setTime.py

gaionim, 2016-09-03:
@michelinok 
aspetta il prossimo merge e riprova.
ora a me funziona anche con ./setTime.py,
ma tieni presente che uso python2,
se il tuo sistema è per python3, probabilmente dovrai lanciare
python2.7 /setTime.py

ora chè anche un abbozzo di help.
Fammi sapere,

F5OEO, 2016-09-03:
Make a merge with your file directly. Could not have time to test it.
Carefull try first with -c 1  !

michelinok, 2016-09-04:
Ok...downloaded again the git, but it seems i cannot pass the --offset parameter ...why?

gaionim, 2016-09-04:
because not implemented yet....
see setTime.py -h

michelinok, 2016-09-04:
Ah ok....maybe i did dowload the script from your git some days ago...and it was implemented there :)
I'll wait when it will be available here.
Many thanks for your work!
'73 de IU5HES   op. Michele

gaionim, 2016-09-04:
Probabilmente è il mio pessimo inglese, era una nota in cui chiedevo di suggerire eventuali altri parametri, ma specificando bene parametri e comportamento atteso,
quello dell'offset era solo un esempio.

Comunque vedo che siamo "vicini di nominativo",
73 de IU2 HDS op. Mauro

michelinok, 2016-10-15:
any news about merging the script? (maybe the script should accept date/time as a parameter)

gaionim, 2016-10-15:
The script is merged and working. 
I feel good with:
 - change the system time
 -  run script to set clock
 -  ntpdate to fix system date

If it is no possible ( because other daemons need the right datetime ) 
write the command line parameter you want, I can try to modify the script.

closed 2018-11-23 by F5OEO