F5OEO/rpitx #38: rpitx installation issue

michelinok, 2016-09-03:
Is it possible that rpitx installation create a conflict with WsprryPi ?
Everything was ok...then i installed rpitx and tryed dcf77 script...no signal on 77.5kHz...and the WsprryPi doesn't transmit anymore (of course...it was closed before rpitx/dcf77).
Should reflash the sd card...
Happened 3 times in a row...
Any idea?

michelinok, 2016-09-03:
tryed right now...4 times in a row...
I'll try to only install rpitx next time (without running dcf77 at all).
Any idea while reflashing the sd card?

F5OEO, 2016-09-04:
If you want to use WsprrPi : 2 solutions.
1- Remove dt-blob.bin in /boot/ and reboot, put dt-blob.bin when use rpitx
2 - recompile wsprpi by changing https://github.com/JamesP6000/WsprryPi/blob/master/wspr.cpp#L58 with #define F_PLLD_CLK   (1000000000.0*(1-2.500e-6))

michelinok, 2016-09-04:
Thanks a lot, will try in some hrs..
Just a little question...should i recompile wsprpi ONLY if rpitx is installed right?

michelinok, 2016-09-04:
modifiyng wspr worked perfectly!
many many thanks!

closed 2016-09-04 by michelinok