F5OEO/rpitx #39: Does not work on OrangePI

georg91, 2016-09-16:
Hello,

today I´ve discovered Rpitx, and wanted to try it out on my Raspberry compatible board: **OrangePI PC**.

I´ve followed the manual for FM Modulation.
When running `sudo ./rpitx -m RF -i fm.ft -f 100000 -l`

Following happens:

```
rpitx Version 0.2 compiled Sep 16 2016 (F5OEO Evariste) running on Master PLL = 1000000000
Wheezy
Failed to open mailbox
MASH 1 Freq PLL# 6
END OF PiTx
root@orangepi:~/rpitx#
```

I guess the error is `Failed to open mailbox` - but I have no idea how to fix this error or which problem causes it. Any ideas?

ha7ilm, 2016-09-16:
Hi @georg91,

Unfortunately, Orange Pi PC is not _totally_ compatible with the Raspberry Pi. It is unlikely that `rpitx` can be easily get working with it.

The Orange Pi PC has an AllWinner H3 SoC on board, while the Raspberry Pi has a Broadcom SoC. 

Software that generate an RF signal on a GPIO pin of the RPi (including pifm, WsprryPi, rpitx, etc.) rely on hardware resources and software interfaces that are unique to the RPi (e.g. they rely on how the DMA works, how PLLs have to be programmed, etc.) 

While it might not be impossible to port some of these software to other ARM SoC based computers, it is not a trivial task.

georg91, 2016-09-16:
@ha7ilm 
Ok I see so I will have to get a raspberry then. :(

`Software that generate an RF signal on a GPIO pin of the RPi (including **pifm**, WsprryPi, rpitx, etc.)`

Pifm was working fine though.

Just to clarify - without additional hardware you can only transmit and not receive?

F5OEO, 2016-09-16:
Interesting question about why Rpitx is only for Raspberry Pi , here is a short explanation 
Unless most of linux softwares which run on differents platforms, rpitx is very closed to Rapsberry Pi and specially Broadcom chips ( BCM2835 et BCM2836).
Why ? 
- Use directly hardware registers (via mailbox)
- Use specific DMA process from Broadcom
- Use PLL very powerfull on Raspberry Pi (not seen a frequency range like that on other chips ..)
- Thanks to a very documented ([https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf]

I have some orange Pi here and "like", but 
- Even documented, PLL is very limited, thus only a litlle range ( few MHZ)
- Community support is not active

Conclusion :

Use Rpi competitors to performs software. As soon as you want GPIO "ehanced modes" (not blinking a led or make I2C communication), forget to dream in compatibility !

F5OEO, 2016-09-16:
"Pifm was working fine though."
Is it ??????? Because if right, I need to investigate (but really doubt)!

ha7ilm, 2016-09-16:
Hi @georg91,

> Pifm was working fine though.

That sounds interesting! Which version?

> Just to clarify - without additional hardware you can only transmit and not receive?

Yes.

georg91, 2016-09-16:
@F5OEO 
I have no in-depth knowledge but this explains it well for me. I didnt consider the very "hardware closeness" of this and that it actually would matter if this is Broadcom or Allwinner, but it makes sense.

And regarding "working" **pifm**

`root@orangepi:~/rpitx# ./pifm tone.wav fm.ft
Reading file : tone.wav
Sample Rate : 48000
Channels    : 1
----------------------------------------
Writing file : fm.ft
root@orangepi:~/rpitx#
`

Of course I cant tell for sure or say what content is actually in fm.ft now. 
But I consider the absence of any error message as "working" in this case :sweat_smile:

F5OEO, 2016-09-16:
Even no error message... Real trust in working is checking if you can receive something through RF ! As seen in the AllWinner documentation, no chance to have a signal up to few MHZ and surely not with the Pifm registers which are specific to Broadcom !

georg91, 2016-09-16:
@F5OEO 
It only generated the file, I couldnt even try to transmit something because rpitx gave me the quoted errormessage.

F5OEO, 2016-09-16:
Not using rpitx but Pifm which seems to work : could you receive a RF signal from you Orange Pi ?

georg91, 2016-09-16:
How to do this? I thought pifm is only for creating the files.

F5OEO, 2016-09-16:
Ah OK.. some confusion.. In rpitx ,Pifm is just a process to convert audio to frequency. Datas are created to be used by rpitx. But this is the same name as "pifm" reference [(https://github.com/rm-hull/pifm)]. Sorry for that "same" name. 
Again and now, be sure : no way to use Orange Pi as a RF transmitter with rpitx (and other raspberry software based)

So

georg91, 2016-09-16:
Even the other pifm doesnt seem to work, it gives no messages and actually "runs" for the lenght of the WAV File but I hear nothing.

closed 2018-11-23 by F5OEO