F5OEO/rpitx #40: Hardware to clean up and amplify signal.

ruant, 2016-09-18:
Hi. 
Kinda new to the raspberry pi and using it to transmit with. 
As I've read the signal generated is pretty nasty so it should be cleaned up if it's going to be let out into the wild.

So is there any hardware out there that does this? 
Preferably with a amplifier as well. 
I did find this QRPi, for the 20M band. But it's not available any more?

I know that this kinda doesn't have anything to do with the software, but some "official" pointers regarding existing hardware/how to build one your self would be great. 
I'm not a hw geek or very good at radio technical stuff, but I know how to use a soldering iron, as many others too probably, if only we had some drawings / guide to follow.

Awesome software, keep up the good work!

michelinok, 2016-09-18:
Wait 10 days.... 😉

ruant, 2016-09-19:
michelinok, cool 👍 
Any sneak peak information? 😄

F5OEO, 2016-09-19:
Yes, lot of request about hardware filtering. Main problem is to adapt of frequency use : filtering is not the same on HF band and VHF/UHF band. Having maybe a switch between several bands should be perfect.
At this time (before the 10 Days of michelinok) there are
- https://shop.languagespy.com/collections/frontpage/products/tiny-lpf-qrp-low-pass-filter-kit
- https://www.tapr.org/kits_20M-wspr-pi.html (But sold out for now)

michelinok, 2016-09-19:
The project I'm working on will take much more time.... Meanwhile you can
order some lpf kits from qrp labs for example

ruant, 2016-09-20:
Thanks for the info guys :)
So is it enough with just a LPF?
The optimal solution I guess would be to use a BPF?

Any pointers regarding amplifying the signal a little bit? 😃

michelinok, 2016-09-20:
I'm using the languagespy kit that consists of a very small amp (150mW) and
an lpf.
Gone as far as 8000km in wspr mode....
Take a look on wspr map and check the IU5HES callsign

ruant, 2016-09-22:
That's pretty far, amazing that such a low power signal can go so far. 

Just to clarify, are you using the kit that F5OEO linked to? https://shop.languagespy.com/collections/frontpage/products/tiny-lpf-qrp-low-pass-filter-kit
Didn't find any info about the amplifier in the specs?

michelinok, 2016-09-23:
The entire kit has been removed from the website some days ago and will be
available in some days again. It consists of an lpf (the one you linked)
and a very very tiny rf amp that you plug onto the raspberry (then you plug
the lpf onto the interface).
You can find some pictures on my website
Http://michelinok76.blogspot.it

> That's pretty far, amazing that such a low power signal can go so far.
>
> Just to clarify, are you using the kit that F5OEO linked to?
https://shop.languagespy.com/collections/frontpage/products/tiny-lpf-qrp-low-pass-filter-kit
> Didn't find any info about the amplifier in the specs?
<https://github.com/F5OEO/rpitx/issues/40#issuecomment-249050699>, or mute
the thread
<https://github.com/

ruant, 2016-11-11:
@michelinok Any news on your solution?

I can't find any solution where there is a small amp included as well.
All is either out of stock or taken of the market due to issues with components/design.

closed 2018-11-23 by F5OEO