F5OEO/rpitx #41: Pi transmitting unwanted signals while transmitting by SSB

chrismatchett, 2016-09-24:
Thanks for rpitx by the way.

I"m doing something like this to transmit a jt65 cq of a pre-recorded file I recorded via ssb on an original raspberry pi.

sudo /home/pi/radio/rpitx/rpitx -m IQ -i /home/pi/radio/rpitx/jt65/cq.wav.ssb -f 14077.6

I can pick myself up on another receiver and decode the jt65 file 20m away. The peculiar thing is my transmit signal is also transmitting a local am or fm radio station alongside the wav file which makes for a right dirty signal. I have a low pass filter for 20m connected between the pi and the antenna.

Has anyone any thoughts on cleaning up this signal? I don't see this behaviour when I use the VFO option as a test.

Many thanks,

Chris

F5OEO, 2016-09-24:
Can you join your cq.wav.ssb file to reproduce here ?

michelinok, 2016-09-24:
How did you generate the wav? I'm really interested!!

chrismatchett, 2016-09-25:
Hi,

I've included a link to the file I am sending, encoded with ssb as per your instructions. I'm pretty confident the file is not the problem as I'm decoding it 20m away. It's the unexplained other signals being transmitted is tripping me up.

As to how I created the file I piped the output of wsjt-x on my ubuntu machine to audacity and took it from there. The only problem is what do if someone replies :-)

http://filebin.ca/2wIk1Eglu6lC/cq.wav.ssb

Chris

F5OEO, 2016-09-25:
Ok, got it.To be sure to speak about same "dirty signal", can you make a snapshot of your spectrum and indicating what is the dirty signal ?

chrismatchett, 2016-09-25:
Here's a look at the spectrum...you can see the signal is filling about 200kHz.

An unwanted station, possibly am/fm, seems to be getting transmitted with the IQ file (I used pissb to make the file). I have a 20m lpf between the pi and antenna...the antenna is  a 5m dipole due to space.

Thanks,

Chris
[image]

F5OEO, 2016-09-25:
Graduation on WSJT-X is in Hz : We see that signal take all the 1500-1700Hz slot, which means 3KHZ (Not 200KHZ).
pissb was optimized for the voice : having a lowpass filter at about 2.5KHZ (standard SSB voice modulation)
This should explain why this spectrum is not "clear" ! 
2 solutions 
- Rewriting the lowpass filter in pissb
- Writing the JT65 using the RF mode which takes frequency/timing (not very difficult) in order to be modulate by rpidatv.

chrismatchett, 2016-09-25:
Thanks for taking the time with this, I know this is not an issue as such.

I'm more clued in now about how your software works and also a bit more clued into radio.

Thanks,

Chris

chrismatchett, 2016-12-05:
Hello again,

I wasn't quite sure how to carry out your suggestions.

> Rewriting the lowpass filter in pissb
> Writing the JT65 using the RF mode which takes frequency/timing (not very difficult) in order to be modulate by rpidatv.

Regarding the second suggestion. Do you mean something along the lines of;

- sudo ./rpitx -m RFA -i filename -f 7.0386

I'm not sure how to get the wav file ready on 6Hz as opposed pissb's 2.5KHz for voice? I've found from last time that 6Hz is the width of the digimode I am trying to send.

Thanks again. My understanding and not your explanation is the problem I'm sure.

Chris

closed 2018-11-23 by F5OEO