F5OEO/rpitx #42: PSK-31

cyber5tar86, 2016-09-26:
Is possible to generate PSK31 signal on GPIO?

F5OEO, 2016-09-26:
It should be possible. For intermediate solution : you can try to use an external software producing a PSK32 wav and then generate an IQ file using pissb. It is not direct and not optimized. 
I think to implement majors digital modes : PSK31, JT65, WPSR soon !

streamofstars, 2016-09-27:
I know this is just a private, open source project so I cannot really expect anything and should just happily use what you already made and published... but perhaps you have a time line set or at least you can give me a rough estimate on new modes implementation? I am interested mostly in WSPR, as I have now Raspberry Pi 3 and Zero, and James' WSPR tool for Raspberry Pi (https://github.com/JamesP6000/WsprryPi) does not work with the newer RPis and his project looks inactive. Other modes such as already mentioned JT65 and PSK31 would be also great.

ha7dcd, 2016-09-27:
Hi,

Something you mistaken here, https://github.com/JamesP6000/WsprryPi is
running nice on RPi 3!
You possibly want to make an install using a fresh image.

73,
HA7DCD
Zoltan

streamofstars, 2016-09-27:
@ha7dcd I was trying yesterday and it did not work - compiled properly, run without errors but my SDR plugged in to another computer 2 meters away did not receive any signal. Rpitx on the other hand, worked flawlessly. So I am not sure what is the issue here. Will remove current installation and try again from scratch later on, if you know for sure WsprryPi works with RPi 3. I have also emailed James about my issues yesterday. Anyway, when it comes to rpitx, it would be great to have all the modes available in one tool, so whether James' implementation works or not, I would still vote for Evariste's rpitx have them implemented.

ha7dcd, 2016-09-29:
You may tryied to wrong GPIO, Wsprry pi works on GPIO4, not the same as
rpitx: http://j.mp/tapr-qrpi

ha7dcd, 2016-09-29:
today tested for you, works nice on my RPi 3: http://imgur.com/C0y7Ufo

michelinok, 2016-09-29:
Look for my other ticket (wssprpi and rpitx conflict or something similar)
You've to recompile wssprpi after a little change on the source code,
otherwise wssprpi won't work anymore after installing rpitx

streamofstars, 2016-09-29:
@michelinok OMG THANKS SO MUCH! That was it! The problem described here https://github.com/F5OEO/rpitx/issues/38 WsprryPi does not transmit when installed along rpitx unless one of the below is implemented (quote):
> 1- Remove dt-blob.bin in /boot/ and reboot, put dt-blob.bin when use rpitx
> 2 - recompile wsprpi by changing https://github.com/JamesP6000/WsprryPi/blob/master/wspr.cpp#L58 with #define F_PLLD_CLK (1000000000.0*(1-2.500e-6))

michelinok, 2016-09-30:
> WsprryPi/blob/master/wspr.cpp#L58 with #define F_PLLD_CLK

I used the second option (modify and recompile)
Glad it helped

G3ZLQ, 2017-01-05:
One more digital mode for rpitx is available. I have created an extension to rpitx for RTTY.
For the time being it can be downloaded from yeovil-arc.com/rpi_rtty.zip
I would much appreciate someone trying it.

streamofstars, 2017-01-06:
@G3ZLQ it seems to work. I tried on 2m. Be advised, your readme_rtty.md file says 
> ./testrtty will send from raspberry pi as described in readme.md
but even after compiling, there is no readme.md file to be found. I see that the message to be encoded is provided in plain text within the testrtty.sh. I have customized, transmitted, received properly but at the end of the transmission I always find a few characters that were not encoded by me. I am not sure if this is just a glitch, my fault, your fault or perhaps you purposely add something at the end.