F5OEO/rpitx #45: Is it possible to stream audio to rpitx and not use wav files?

antonjan, 2016-10-13:
Is it possible to stream IQ signal demodulated from rtl_fm or possibly audio from microphone to rpitx?

ha7ilm, 2016-10-13:
@antonjan See: https://github.com/ha7ilm/rpitx-app-note

OutsourcedGuru, 2017-05-18:
For what it's worth, I've got this working to stream from a USB-based microphone, noting that mine shows up on **plughw:1** to transmit mono on FM 87.9 Mhz:

```
arecord -D plughw:1 -f S16_LE -r 48000 /dev/stdout 2> /dev/null | ./pifm /dev/stdin /dev/stdout | sudo ./rpitx -m RF -i /dev/stdin -f 87900
```

F5OEO, 2017-05-18:
Use sox between arecord and pifm should work to resample and make it mono. In a short term (working on dev branch, this would not been required anymore).