F5OEO/rpitx #46: Possible to capture and transmit RF from remotes

mikenolet, 2016-10-22:
I have several RF remotes that control various devices (ceiling fan, fire place, blinds) and would like to capture the RF transmission that is sent when a button is pressed so I can capture and incorporate into a home automation solution I use.  Wondering if this is possible with RPITX.

redfast00, 2016-10-22:
@mikenolet: I'm afraid this is not possible with rpitx since it can only transmit and not receive. Depending on the frequency the remotes transmit on, this may be possible with an RTL-SDR.

F5OEO, 2016-10-26:
Remember that rpitx has not a huge bandwidth..so in order to work, you have to record your signal of interest in I/Q after decimation in order to have a low bandwidth no more than 250KSymbols.