F5OEO/rpitx #50: Possible to send at 868MHz?

maciboy, 2016-11-08:
Hey,
just saw this awesome project and got an idea for my home automatization:
I have a „Hörmann“ garage door opener.
For a long time, I want to use my phone instead of the original sender.
Therefore, I thought to run a raspberry and send the signal from there.
Until now, I had no idea how to make the raspberry send, but this repository seems to have the ideal solution :)

Unfortunately, I found out, that my garage door opener works at 868MHz, which seems to be not supported  by rptix. (http://www.hoermann-faq.com/antriebe-funk/welche-sendefrequenz-nutzt-mein-handsender/)
Is it true that 868MHz will not work? Why is limit at 750MHz?
Is there the possibility to make it work? That would be great!

Thank you so much in advance!

mikenolet, 2016-11-08:
Believe the Pi can only transmit up to 500 Mhz

maciboy, 2016-11-08:
Mmh, I think the 500MHz are a bit outdated, since in the repository description ist already says:
„-f float    frequency to output on GPIO_18 pin 12 in khz : (130 kHz to 750 MHz)“
And 750MHz is not too far away from 868MHz so I’m still hoping ;)

mikenolet, 2016-11-08:
I am trying to do the same sort of thing.  Were you able to capture the source and create an IQ file?

emard, 2016-12-01:
RPi actually transmits square wave and it has strong 3rd harmonic,
so one way to get to 867.84 MHz is to transmit at 289.92 MHz and make
a precise lambda/4 antenna that self-resonates at 868 MHz
this is simple wire about 86 mm long, also the PCB and pin wiring
is included in this length so the actual wire for RPi GPIO18 antenna should 
be shorter, and you can experiment by cutting it off mm by mm until it works best

jnk0le, 2016-12-03:
It could be possible to use 5th harmonic, but you should use band pass filter and some kind of PA to get it working. 
Of course it's possible only if pi can output 868Mhz harmonic on its pin.

maciboy, 2017-01-08:
Thanks!
Using the testfm.sh sending at 289.42 MHz works.
I can receive a weak signal using HackRF and GQRX at 868.283 MHz 👍 

I also have recorded the signal of my garage door opener using HackRF. So I have a .bin-file, I hope it’s possible to send the data in there using rpitx. As far as I can see, data must be in .iq or .ft format, not sure how to get it in this format at the moment…

emard, 2017-01-09:
Great ;) 3rd harmonic has 30% amplitude relative to 1st harmonic
so the output power is quite useable. RPI clock oscillator is not thermally
stable. Let it warm up properly and then re-tune the sender freq to improve
the range

maciboy, 2017-01-09:
Another question ;)
Why can’t I use 434.14 MHz (2nd harmonic)? Shouldn’t be the signal strength better at that frequency?
If so, what would be the optimal length of the antenna?
(I calculated the 86mm for a λ/4 antenna back to the frequency and ended at approx. 871 MHz, so I can’t see how the 289.42 MHz play into that and what adaptions need to be made for 434.14 MHz)

A more general question:
as I said, I recorded a .bin-file using hackRF. Samplingrate is 1MHz. (hackrf_transfer -r door.bin -f 868000000 -s 1000000)
How do I have to transform it, so I can send it via rpitx? Is this .bin already the right format to use „-m IQFLOAT“ (~raw). Would be nice if sb could tell me, how to change the format or which mode is the correct one :)

Thanks so much!

emard, 2017-01-10:
You can't do 2x freq because digital pins output square
wave which has only odd harmonics, 1, 3, 5, no even's
like 2, 4,... I'm sorry :).

https://en.wikipedia.org/wiki/Square_wave

On second question about hackrf/rpitx file conversion
formats I don't know the answer :)

maciboy, 2017-01-26:
Unfortunately I’m stuck :(
I can’t get rpitx to send a signal that opens my Garage door…

Using HackRF I can see that rpitx sends something out, but it looks weak, noisy and also takes much longer to send than the original record is…
I figured out, that HackRF seems to be able to record .wav files, but using them doesn’t work the way I want…
I have tried several records (.bin, .wav), different modes of rpitx, played around with the sampling rate.
Had no luck, I think I’m doing something wrong.
Any ideas or tips?

ha7ilm, 2017-01-27:
Some very-very general ideas for signal replay:
* You could inspect the waveform contained in the file. You could use e.g. Audacity.
* The amplitude of the signal should be normalized.
* The sampling rate and the format should match the tool. 
* `rpitx` tools don't need a WAV file header as far as I remember.

maciboy, 2017-02-10:
Today is the day when my garage door reacted to the Pi’s signal 🎉 

I **recorded the signal with HackRF One** as .wav, the problem here is that HackRF records with a sampling rate much higher than 48kHz (by default it is 10MHz!).
So one has to **downsample the recorded file**, which I did with Matlab (downsample(signal, „fs_of_record/fs_for_rpitx“)).
The resulting .wav signal is now send at 433MHz (either my garage door listens their too, or 2nd harmonic works?!) as **IQ-mode with '-c 1'**.

I’m so happy right now, this took me so many hours 💃 
Thanks @F5OEO for this tool!

F5OEO, 2017-02-10:
Very happy that you succeed in ! All hours are not lost as I hope that you learn many things. Doing downsample with mathlab is OK, should have some more light tools but all the ways are allowed !
Sorry I was not not always very helpfull. Like that you make a story of your experiment to share with others.

maciboy, 2017-02-10:
Thanks for your kind reply @F5OEO 😃 
I have one, hopefully last question, before the whole setup works…

Until now I used a script similar to the ones you provided (Garage.sh, stored in the 'rpitx' folder):
_sudo ./rpitx -m IQ -i signal-short.wav -f 433000 -c 1_

When double click this file, I’m asked if I want to open or execute the script. If I click execute, then the garage opens.

Now, since everything works, I want to automate the call of the script.
(For this I use 'Homebridge' to get a button for the garage door on my iPhone, which starts the script. This is just an info…)
So my question is now, how to call the „Garage.sh“ script for execution?
I extended the script like this:

_#!/bin/bash
sudo /home/pi/rpitx/rpitx -m IQ -i signal-short.wav -f 433000 -c 1_

If I now try to run the script in terminal with „/home/pi/rpitx/Garage.sh“, I get the following error:

_pi@rasp:~ $ /home/pi/rpitx/Garage.sh
Use GPCLK Pin instead of PWM
Failed to read Filein signal.wav
END OF PiTx_

I looked into your code and found the lines that are responsible: (https://github.com/F5OEO/rpitx/blob/master/src/RpiTx.c)
…
case 'c': // Use clock instead of PWM pin
			UsePCMClk = atoi(optarg);
			if(UsePCMClk==1) printf("Use GPCLK Pin instead of PWM\n");
			break;
…

What does the error mean and why does it appear?
Why can I execute the script manually without this error but can’t call it with terminal? 🤦‍♂️ 
Or in general, what would you suppose to do, if I just want to start the transmission/script from another program.

Since this is the last step needed to open my garage door using my phone, I would be so glad, if this could be solved 😃

bend94, 2017-06-19:
Hi 
I want to do the same but with my Thermostat which use 434Mhz.
That is to say, capture the Wav signal 434Mhz (i did it already with SDR sub and GQRX)
I convert the wav  to mono.

Now i want to send it with rpitx. one question , do i have to connect just a wire to the GPIO pin ?
What length for the wire ? 
Which Gpio pin ?

Regards.

maciboy, 2017-06-22:
Hey :)
Yep, it only needs a small wire!
I connected it using the „-c 1  (Transmit on GPIO 4 (Pin 7) instead of GPIO 18)“ option, since I experienced better transmission using this one…
Length of the wire is about 8cm, as proposed by @emard above.

Good luck!

maciboy, 2018-11-24:
@F5OEO 
Hi :)
What do you mean with „Replay example in V2“?

F5OEO, 2018-11-24:
Update rpitx with latest master version : Version 2. You have an example for recording and replaying a signal by launching rtlmenu.sh. Replay on 868MHZ is not in the frequency range, however, you could use harmonic 3 for example and transmit on 289.333Mhz.

Commander-Crash, 2019-07-28:
I know this is old but I use the old ver of rpitx. 868Mhz transmission can be achieved if you use a small cheep amp like the one here https://images-na.ssl-images-amazon.com/images/I/511pQDuhngL._SX342_.jpg A word of warning is that you may cause your pi to become unstable after long periods of use and if your PSU is not rated for at least 2.5A, and not to mention that the harmonics is all over the place. Use caution.