F5OEO/rpitx #51: how to transmit in am mode?

kensukekita, 2016-11-28:
how to transmit in am mode?
any idea?

orrinwinton, 2016-12-02:
Same question here. I am able to use testfm.sh okay, but testam.sh and testssb.sh give muffled or distorted audio on my receiver set to AM or USB.

kensukekita, 2016-12-02:
Yeah same here :'(

silicator, 2016-12-02:
dito here! and how is it with (pmr) ctcss-tones/sub-channels?

hirve, 2016-12-21:
Why the modulation in AM mode is so high?
But when I use the wav file with reduced volume, the sound is distorted. When I use the wav file with very low volume, I get silence. Sounds like there is a some threshold. DC shift made by modification **piam.c** file allows to get the carrier but can't eliminate distorsion.

erm2587, 2017-08-06:
Hi,

First of all, thank you @F5OEO for sharing your work.

I'm using the AM modulator for testing/repairing a 1945 valve radio. I'm able to tune the signal (1000 Khz) but I'm getting a lot of distortion and I'm not sure if it is related with the radio or the way the AM signal is been generated. Could you help me?

Regards

orrinwinton, 2017-08-06:
Elías, the only mode i have been able to get success with, is FM. AM and
SSB i get distortion.

Orrin wn1z
NE Calif

erm2587, 2017-08-06:
Thank you for your feedback @orrinwinton.
Lets see if I have time to take a look to the code, I'll let you know if I find something.