F5OEO/rpitx #52: DCF77 setTime.py 1 second lasts 1.2 seconds

emard, 2016-11-30:
On rpi2 and rpi3 I have problem with setTime.py

I connected GPIO18-GND to series resonant LC antenna, 
Carrier is great, 77.5 kHz precise to 4 decimals, and can be 
cleanly received by the clock few meters away.

the DCF time binary message is correct but the bits transmitted have wrong timing.

everything is slowed down by about 20% so the time distance between 2 adjacent
1-second bits (pulses) actually last about 1.2 seconds, short pause 120ms is too long 
(should be 100ms), long pause 240ms is too long, so it clock won't accept it

Should I somehow use dt-blob.bin? 
(77.5 kHz transmitter uses 19.2 MHz base, I understood the blob matters for 
high frequencies which need clock base 1 GHz)

Any help

gaionim, 2016-12-04:
How about UP time ?  It is 20% longer too ?
If so I can add a calibration parameter.

michelinok, 2016-12-04:
It would be great if you could add a parameter to specify a specific
date/time to set
'73 de IU5HES

=================
Http://www.iu5hes.it
=================

F5OEO, 2017-05-17:
Hope that dev branch commit solve this timing issue.

emard, 2017-05-17:
Hey thanx I can dust off my setup and try again with the fresh update!