F5OEO/rpitx #53: IQ samples from TCP/UDP

bssstudio, 2016-12-07:
I would just like to discuss if it would be possible to extend the input options to allow receiving raw/float IQ samples from the network (using either TCP or UDP socket). That would extend the usability of this project as one could then prepare the TX chain in (for example) GNU Radio and output it to the RPiTX to transmit via network.

Comments greatly appreciated.

F5OEO, 2016-12-07:
Yes, rpitx will be based on a  library soon : thus could be inserted in SoapySDR or Gnuradio. IP could be added more easy then. For now it could already be done but not directly (see notes from HA7ILM). I mainly work on rpidatv for now, but back soon to improve rpitx. Thanks for supporting it !

ha7ilm, 2016-12-07:
@bssstudio see: https://github.com/ha7ilm/rpitx-app-note

closed 2017-05-17 by F5OEO