F5OEO/rpitx #54: What output Impedance could I expect from Raspberry 2B or 3B?

antonjan, 2016-12-12:
What output Impedance could I expect from Raspberry when I am transmitting with FM on VHF frequency 145Mhz (Ohm)?

F5OEO, 2017-05-17:
That's a difficult question. It seems that output is around 300 ohm, but depend on frequency and dynamic in amplitude (not used in FM). Help required to better know impedance !

F5OEO, 2018-05-07:
Thanks for this very nice contribution to understand better GPIO impedance
when it is used in RF. Maybe could be reproduced using a VNA ?

felixzero, 2018-05-07:
Unfortunately I don't own such equipment. These can cost as much as a car...

kj7rrv, 2021-06-10:
@felixzero have you heard of NanoVNA? They seem to be a bit more expensive in Europe than in the US, but it looks like you can get one at https://www.passion-radio.com/meter/nano-vna-918.html?search_query=nanoVNA&results=6#/221-screen_size-2_8_7_11cm for €65.
73 de @kj7rrv