F5OEO/rpitx #58: AM signal replay

whit98c, 2017-01-07:
I've captured some AM ASK/OOK signals with an SDR and Audacity. What do I need to do to get these to play back on the same frequency as AM?

I'm trying this and I see a signal in SDR# but it's not sending antyhing close to the original signal. Maybe I'm going about this the wrong way?

sox office.wav os.wav rate 48000 channels1
./piam os.wav am.rfa
sudo ./rpitx -m RFA -i am.rfa -f 314981 -l -c 1

Maybe it's the file that's the problem? Here is the WAV file
https://www.dropbox.com/s/dup38l3fml7yy7k/3passes.wav?dl=0

Another approach is I have the bit sequence and timing (I think), if there's a way to create a data file manually using that information.

hirve, 2017-01-11:
I tried much playing AM. Seems AM and SSB are emulated with 1 bit keying, so it's a little problem to get perfect signal.

whit98c, 2017-01-20:
Thanks. I ended up using a 315Mhz transmitter to get the job done.

F5OEO, 2017-05-03:
Prefere to record a I/Q decimate signal and playing it with -IQ mode instead.