F5OEO/rpitx #59: Problem with harmonic mode?

jlc66, 2017-01-12:
Perhaps I am misunderstanding something, in which case I apologize.

If I do, e.g.,

`sudo rpitx -m VFO -f 720000`

I see

` Warning : Using harmonic 3`

and

`WaitNano=25000 F1=159993750.244131 TuneFrequency 160000000.000000 F2=160000000.000000 Initial Resolution(Hz)=6249.755869 ResolutionPWMF 41.944670 NbStep=149 DELAYStep=9`

Why is TuneFrequency 160 MHz?  I expected 720 MHz / 3 * 2 = 480 MHz, or three times what I see, which would produce a 240 MHz fundamental and a 720 MHz third harmonic.

I do indeed see signal at 80 MHz.  From the code, it does look like the frequency gets divided by 3 twice: in pitx_SetTuneFrequency there is this:

`GlobalTuningFrequency=Frequency/HarmonicNumber;`

and then in pitx_run FrequencyAmplitudeToRegister is called with GlobalTuningFrequency//HarmonicNumber.  So it looks like my desired frequency is the 9th harmonic rather than the 3rd.

Again, apologies if I have misunderstood.

F5OEO, 2017-01-12:
Here is a more detailed explanation : 

By default, modulation output is done with PWM engine. It outputs a 32bit pattern which is mastered by the clock. The pattern is 101010...in order to simulate a cycle. It means that it outputs a frequency/2 (compared to clock). That's explain the /2.

There is an other mode (prefered) which use directly the clock output :add  -c 1 to the command line and use GPIO 4 (Pin 7 of header). Signal is not divided by 2 and is cleaner than with PWM.

Hope that it answers your question. Feel free to ask other details if needed.

jlc66, 2017-01-12:
I understand the factor of 2, which is actually a multiplication by 2.  I am saying that instead of dividing by 3 when using the 3rd harmonic, the program appears to divide by 3^2 = 9.  For example, when 720 MHz is requested, rather than generating a 240 MHz square wave, it generates an 80 MHz square wave.  This is due to dividing by the HarmonicNumber in twice.

F5OEO, 2017-01-13:
Yes there is something weird..Have to check the code , something suspect in calculating. Thx for pointing this bug.

closed 2017-05-17 by F5OEO