F5OEO/rpitx #60: Terrible audio in AM, also PiFmRds won't work if you install it!

pilovis, 2017-01-18:
Very bad program!
The AM modulation is so awful.
I'm trying to understand how to recover my fantastic PiFmRds :(

pilovis, 2017-01-18:
Weak programmers do not write instructions how to uninstall their programs.

To remove rpitx and to get PiFmRds working again, follow these steps:

rm /boot/dt-blob.bin
rm /usr/bin/pifm
rm /usr/bin/piam
rm /usr/bin/pisstv
rm /usr/bin/pissb
rm /usr/bin/pifsq
rm /usr/bin/rpitx

then go to "PiFmRds/src" and run "make clean" and "make" to reinstall PiFmRds, then reboot.

nexus166, 2017-01-18:
pro users make backups

pilovis, 2017-01-18:
The worst is that  "/boot/dt-blob.bin" changes the default pin configurations of your Raspberry (persistent modification after reboot)!

At least he should have warned us before starting the installation of this program.

F5OEO, 2017-01-19:
Warning : Read the GPL Licence :

> This program is distributed in the hope that it will be useful,
>     but WITHOUT ANY WARRANTY; without even the implied warranty of
>     MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
>     GNU General Public License for more details.

closed 2017-01-19 by F5OEO