F5OEO/rpitx #62: is it possible to use another GPIO pin ?

isfanl, 2017-02-13:
i recently working on a project (actually an assignment from my lecturer). i tried to make my raspberry pi 3 works with a 5" touchscreen, unfortunately the touchscreen requires all the first 26 gpio pins, in other word it covers the pin 12 that supposed to be connected to a wire antenna. so is there any possible way to use another pin for the output signal ? and how to do that

F5OEO, 2017-02-13:
This is not PIN 12 but on GPIO 12 : means PIN 32 of header -> not in the first 26 first pins. So it should work.

isfanl, 2017-02-14:
[image]
i meant this, the touchscreen requires all the pins in the blue area, meanwhile the antenna connected to gpio 18 that in the red area, 
what code should i add to the script to use another gpio pin (outside the blue area) ?
thank you very much for your respond ^_^

F5OEO, 2017-05-17:
There is for now 2 ways of output : 
GPIO 12 means pin 32 of header 
OR
GPIO 4 means pin7 of header

streamofstars, 2017-08-19:
> There is for now 2 ways of output :
> GPIO 12 means pin 32 of header
> OR
> GPIO 4 means pin7 of header

I am puzzled now, in the readme you mention "Plug a wire on GPIO 18, means Pin 12 of the GPIO header (header P1)" and "-c 1 Transmit on GPIO 4 (Pin 7) instead of GPIO 18". So we know for sure we can transmit on:

- GPIO 18 - pin 12 (default)
- GPIO 4 - pin 7 (with -c 1 switch)

Can we or can't we also use GPIO 12 - pin 32, and if so, how to enable it?

closed 2018-11-23 by F5OEO