F5OEO/rpitx #64: Transmitting AM from RFA is confusing/badly documented

jayslovak, 2017-02-21:
I was trying to transmit a simple AM modulated signal by writing the RFA file from my custom C program.

The file format is described in the help:

```
-m            {IQ(FileInput is a Stereo Wav contains I on left Channel, Q on right channel)}
              {IQFLOAT(FileInput is a Raw float interlaced I,Q)}
              {RF(FileInput is a (double)Frequency,Time in nanoseconds}
              {RFA(FileInput is a (double)Frequency,(int)Time in nanoseconds,(float)Amplitude}
              {VFO (constant frequency)}
```
There are no further definitions on what are the ranges for these values. By looking at the sample files and code of various rpitx programs I was able to deduce the following:

- RF and RFA file format is identical, ie. even the RF file has the amplitude field "(double) frequency; (int) time; **(float) amplitude**;", at least it looks that way
- RFA does NOT use the amplitude field to store the amplitude value, it instead hijacks the frequency field for this purpose. Here is a snippet from [piam.c](https://github.com/F5OEO/rpitx/blob/master/am/piam.c):
```
typedef struct {
		double Frequency;
		uint32_t WaitForThisSample;
} samplerf_t;
samplerf_t RfSample;
```
- The only reason the file has a (float) sized field in it is because the structure above needs to be properly aligned to the 64bit memory boundary, hence the compiler enlarges structure size to 16B instead of 12B.
`readWrapper(&SampleRf,sizeof(samplerf_t)); \* sizeof() will return 16, not 12! *\`
- Finally the amplitude values should be in the range <0 , 32767>, anything in between is emulated by OOK.

Please document how AM mode is implemented and used, I have spent two good evenings trying to understand all of this :-)
Also the struct holding frequency and delay will likely produce incompatible files on 32bit and Big Endian CPUs...

Thanks!

F5OEO, 2017-02-21:
All your comments are right and pleased that you have done such a code analysis. Sorry for missing documentation and confusing. With you permission I will added in the documentation. Hope also to be more active on the project (actually mainly work on rpidatv).

If you have custom software generating other  kind of modulation, welcome to contribute to the project.

Sorry again for your evening's headache :)

jayslovak, 2017-02-21:
Excellent, yes please do.

My goal is to generate AM signal to control garage doors over 433Mhz. I may consider submitting my code once it looks presentable.

BTW here are example signals:

Original:[image]
rpitx:[image]

The signal looks awful due to lacking hardware on the raspberry pi (rpitx needs to tune up and down to approximate the frequency), but this may be good enough for the receiver, more testing needed.

closed 2018-11-23 by F5OEO