F5OEO/rpitx #65: rpitx process always running in the background

maciboy, 2017-04-27:
Hi,
some time ago I wrote about opening my garage door with rpitx… (https://github.com/F5OEO/rpitx/issues/50)

Now, for several weeks the Raspberry Pi (3) lives in my garage, running 24/7.
As I wrote in the other thread, a triggered script runs the rpitx command when I want to open the door.

However, it seems as if the rpitx process isn’t properly ended when transmission ends, since the process is running on and on in the background and so the Pi consumes much more power, as it would in a real idle state…

[image]

What can I do about this?
Is it a bug or is the rpitx process supposed to run forever when started once?

Thanks so much in advance!

F5OEO, 2017-04-27:
Can you put the script to inspect. rpitx should stop after sending (until you not specify -l for looping)

maciboy, 2017-04-27:
Here it is:

_#!/bin/bash
sudo /home/pi/rpitx/rpitx -m IQ -i /home/pi/rpitx/signal-short.wav -f 434000 -c 1
echo „Garagentor oeffnet."_

Sending stops correctly, so there is no looping, just the process seems to be not terminated correctly :/
Thanks for the quick reply!

F5OEO, 2017-05-03:
Can you test with the dev branch just commited.. Don't see the issue here.

maciboy, 2017-05-06:
Hi,
today I tried to install the dev-branch.
Unfortunately, installation was not successful, giving several errors and sending in the end does not work :(

This is the output of install.sh:

`…
libsndfile1-dev is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
imagemagick is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
mknod: ‘/dev/rpidatv-mb’: File exists
cc -Wall -g -O2 -Wno-unused-variable  -o ../rpitx  RpiTx.c RpiGpio.c mailbox.c RpiDma.c raspberry_pi_revision.c -lm -lrt -lpthread  
cc -Wall -g -O2 -Wno-unused-variable -o ../pissb ../ssbgen/ssb_gen.c ../ssbgen/test_ssb.c -lm -lrt -lpthread -lsndfile 
cc -Wall -g -O2 -Wno-unused-variable -o ../pisstv ../sstv/pisstv.c  -lm -lrt -lpthread  
cc  -o ../pifsq ../fsq/pifsq.c   
RpiTx.c: In function ‘SetupGpioClock’:
RpiTx.c:235:14: error: ‘GPPUD’ undeclared (first use in this function)
     gpio_reg[GPPUD]=PULL_OFF;
              ^
RpiTx.c:235:14: note: each undeclared identifier is reported only once for each function it appears in
RpiTx.c:237:14: error: ‘GPPUDCLK0’ undeclared (first use in this function)
     gpio_reg[GPPUDCLK0]=(1<<4); //GPIO CLK is GPIO 4
              ^
RpiTx.c: In function ‘CalibrateSystem’:
RpiTx.c:813:9: warning: variable ‘hFileCsv’ set but not used [-Wunused-but-set-variable]
     int hFileCsv;
         ^
cc -Wall -g -O2 -Wno-unused-variable -o ../pifm ../fm/pifm.c  -lm -lrt -lpthread -lsndfile 
cc -Wall -g -O2 -Wno-unused-variable -o ../piam ../am/piam.c  -lm -lrt -lpthread -lsndfile 
cc -Wall -g -O2 -Wno-unused-variable -o ../pidcf77 ../dcf77/pidcf77.c  -lm -lrt -lpthread -lsndfile 
cc -g -O2 -Wno-unused-variable -o ../piopera ../opera/opera.c  -lm  
Makefile:9: recipe for target '../rpitx' failed
make: *** [../rpitx] Error 1
make: *** Waiting for unfinished jobs....
cc -Wall -g -O2 -Wno-unused-variable  -o ../rpitx  RpiTx.c RpiGpio.c mailbox.c RpiDma.c raspberry_pi_revision.c -lm -lrt -lpthread  
RpiTx.c: In function ‘SetupGpioClock’:
RpiTx.c:235:14: error: ‘GPPUD’ undeclared (first use in this function)
     gpio_reg[GPPUD]=PULL_OFF;
              ^
RpiTx.c:235:14: note: each undeclared identifier is reported only once for each function it appears in
RpiTx.c:237:14: error: ‘GPPUDCLK0’ undeclared (first use in this function)
     gpio_reg[GPPUDCLK0]=(1<<4); //GPIO CLK is GPIO 4
              ^
RpiTx.c: In function ‘CalibrateSystem’:
RpiTx.c:813:9: warning: variable ‘hFileCsv’ set but not used [-Wunused-but-set-variable]
     int hFileCsv;
         ^
Makefile:9: recipe for target '../rpitx' failed
make: *** [../rpitx] Error 1
Installation done
`

(I copied the old rpitx-folder before and everything still works there, so no need to panic. I just wanted to inform you…)

F5OEO, 2017-05-17:
Could not reproduce the hang issue here. Maybe size of IQ file could be an issue. Can you send me your signal-short.wav (promise I can't reach your garage door to open - but you can mail it for confidence) !

maciboy, 2017-05-17:
Funny thing is that this issue never reappeared since I opened this issue here 😄
Before it happened several times, but now everything works normal…
I really don’t understand, maybe it was an issue with Raspbian?!
However, thanks again for caring!

closed 2018-11-23 by F5OEO