F5OEO/rpitx #66: AMs

dimpap5555, 2017-05-08:
There is no tutorial showing how to make am signals
Also i try to convert a wav file and it say me tha it is not 48k sample rate WHY THERE IS NO HELP

ha7ilm, 2017-05-08:
@dimpap5555 Chill out. This is open source software. If you do not find the available help sufficient, you should add it yourself and create a pull request on GitHub. 

You can resample your WAV file using the `sox` command to 48 ksps.

If you don't succeed anyway, you can try to use this guide as well:
https://github.com/ha7ilm/rpitx-app-note#modulate-from-raw-audio-file
However, you'll need a 48k sampled raw file (without the WAV headers) for that.

dimpap5555, 2017-05-08:
Sorry for being very angry but I am new to linux and i am used to sites
with tutorial
Thank you

tnadolecki, 2017-05-08:
Has nothing to do with site. Lack of documentation serves as extra protection. You need to know what you're doing if you want to transmit...

F5OEO, 2017-05-17:
@tnadolecki Lack of documentation is not an extra protection. This an "engine" for transmission, you could read @ha7ilm application note for higher level.