F5OEO/rpitx #68: FM transmission seems to lower the pitch

OutsourcedGuru, 2017-05-18:
Was wondering if anyone has noticed this, too.  My own voice and that of the author appears to be lower in pitch than a true rendering.  I can also transmit realtime from microphone music that's played on my iPhone and via the FM receiver's speaker things also appear to be lowered slightly in pitch.  So, Ed Sheeran sounds like he's grown up a little, for example.

Raspberry Pi 3 Model B, fully-updated Raspbian Jessie Lite and the latest rpitx:
```
arecord -D plughw:1 -f S16_LE -r 48000 /dev/stdout 2> /dev/null | ./pifm /dev/stdin /dev/stdout | sudo ./rpitx -m RF -i /dev/stdin -f 87900
```

I can always hack at your code but it would be nice to know where I might look.  I'd guess /src/RpiTx.c but nothing looks promising so far.  I tried using the `-p float` argument for frequency correction but it didn't seem to affect things in any way that I could determine; I'm guessing that this adjusts the transmitted frequency rather than the audio itself.

I played with the `-d int` DMABurstSize but that also didn't seem to affect the outcome.

I'm guessing that this happens in transmission rather than on the recording or converting steps.

Any thoughts to the best place within the code to start hacking?  Thanks.

F5OEO, 2017-05-19:
Please checkout the dev branch. Fixing some timing issues which cause this kind of pitch problem

OutsourcedGuru, 2017-05-19:
Will do.  (Thanks)

OutsourcedGuru, 2017-05-21:
Interesting.  Somewhere the `dev` branch appears to be addressing the pitch-related problem nicely.  The pitch of my own recorded voice or from previous records and even that of Ed Sheeran appears to be on-pitch.  Well done.

And yet, this particular branch at the moment seems to crash the pi on Ctrl-C at the end.  It becomes unresponsive then to `ping`, other attempts to `ssh` into it, etc.  I've tried various other modes to include just playing a ten-second recorded `.ft` file and even though it might return to a shell prompt you get nothing after that.  So I can't even run a `ps -ax` to see what's running.  :(

Anyway, it was good to hear the proper pitch from the `dev` branch.  I'll next do a compare and to see what changes you were attempting.  (Thanks.)

Unfortunately, now the previous folder's install from the master branch no longer works.  It just pushes static.

For what it's worth, here is what I did:

1. In the Raspi's ~/Sites folder the first time, I did `git clone https://github.com/F5OEO/rpitx.git` into a ~/Sites/rpitx folder and ran the `./install.sh` script in this folder.
2. Recently, in the ~/Sites folder this time, I did `git clone -b dev https://github.com/F5OEO/rpitx.git rpitx.dev` into a new folder and ran `./install.sh` there to test.
3. Note then that the ~/Sites/rpitx then stopped working as expected.  It only transmitted static.
4. In the ~/Sites/rpitx folder, again I ran the `./install.sh` script.
5. The code in this folder still doesn't work, transmitting static.

```
pi@pi3:~/Sites/rpitx $ sudo ./rpitx -m RF -i files/count.ft -f 87900 -l
rpitx Version 0.2 compiled May  6 2017 (F5OEO Evariste) running on Master PLL = 1000000000
memory: 1024 MB
processor: Broadcom BCM2837
i2cDevice: /dev/i2c-1
model: Model B Pi 3
manufacturer: Embest
pcb revision: 2
warranty void: no
revision: a22082
peripheral base: 0x3f000000
Jessie
Using mbox device /dev/vcio.
3616000 Size NUM PAGES 883 PAGE_SIZE 4096
MASH 1 Freq PLL# 6
Error: NTP
WaitNano=14580 F1=175793991.416309 TuneFrequency 175800596.191406 F2=175801536.546633 Initial Resolution(Hz)=7545.130324 ResolutionPWMF 82.012286 NbStep=92 DELAYStep=7
****** STARTING TRANSMIT ********
^CEND OF PiTx
```