F5OEO/rpitx #69: The hyphen versions for the filenames when chaining commands doesn't work

OutsourcedGuru, 2017-05-18:
For what it's worth, I'm just documenting what I found.  The first two commands (arecord + pifm) _when daisy-chained using the hyphen version of this_ seem to work.  It's when you try to add the third command (rpitx) into the pipeline that it just produces static.

Substituting the device names `/dev/stdout` and `/dev/stdin` throughout works as expected.  You could just mention this in the documentation rather than focusing any effort on a fix.

Works:
```
arecord -D plughw:1 -f S16_LE -r 48000 /dev/stdout | ./pifm /dev/stdin /dev/stdout | sudo ./rpitx -m RF -i /dev/stdin -f 87900
```

Doesn't work:
```
arecord -D plughw:1 -f S16_LE -r 48000 - | ./pifm - - | sudo ./rpitx -m RF -i - -f 87900
```

Best version (without the overrun warnings):
```
arecord -D plughw:1 -f S16_LE -r 48000 /dev/stdout 2> /dev/null | ./pifm /dev/stdin /dev/stdout | sudo ./rpitx -m RF -i /dev/stdin -f 87900
```

F5OEO, 2017-05-19:
Thx for comment. Working on a new way of modulation thus will be fix at the same time.

OutsourcedGuru, 2017-05-19:
No problem (and thanks for an awesome bit of code).

closed 2017-05-19 by OutsourcedGuru