F5OEO/rpitx #73: Unable to determine hardware version

UgoPoddine, 2017-06-20:
Hello everyone,

after having updated the RPI system one month ago, I'm currently getting the following error when starting rpitx :

Unable to determine hardware version. I see: Hardware : BCM2835 / expecting BCM2708 or BCM2709_

Previously rpitx ran perfectly. The issue happens both on PI2B+ and on PI3.
The current system version is :  4.9.20+ #985

Any help will be appreciated
73, IU1IPB Ugo

F5OEO, 2017-06-20:
HI Ugo,

Thx in pointing this issue. Hardware version is handle by https://github.com/AndrewFromMelbourne/raspberry_pi_revision which I didn't update for a while. Need to check if he had made correction from latest 4.9 kernel version.

Note that current work is under /dev branch on rpitx.

UgoPoddine, 2017-07-29:
Hello Evariste,

were you able to solve this problem ? Probably it's better if I do a downgrade of the RPI kernal version, in the meantime, isnt'it ?

Thank-you, regards
Ugo