F5OEO/rpitx #74: AIS (GMSK modulation)

sandman7920, 2017-07-19:
Is it possible to use rpitx for transmitting recorded AIS wav file.

> An AIS transceiver normally works in an autonomous and continuous mode, regardless of whether it is operating in the open seas or coastal or inland areas. AIS transceivers use two different frequencies, VHF maritime channels 87B (161.975 MHz) and 88B (162.025 MHz), and use 9.6 kbit/s Gaussian minimum shift keying (GMSK) modulation over 25 kHz channels using the High-level Data Link Control (HDLC) packet protocol. Although only one radio channel is necessary, each station transmits and receives over two radio channels to avoid interference problems, and to allow channels to be shifted without communications loss from other ships. The system provides for automatic contention resolution between itself and other stations, and communications integrity is maintained even in overload situations.
[https://en.wikipedia.org/wiki/Automatic_identification_system](url)

F5OEO, 2017-07-19:
It should be possible to modulate GMSK with rpitx. This is on first tasks I have on TODO list (for other application) . But not sure transmitting AIS with rpitx is barely legal !

sandman7920, 2017-07-19:
This is only for test (indoor).
rpitx transmit
https://github.com/dgiardini/rtl-ais receive

F5OEO, 2017-07-19:
Understand for local testing . As rpitx is I/Q input, you could help me implement GMSK if you like. I figure to use efficient liquid-dsp to do that. Great community work on AIS decoder : congrats !

sandman7920, 2017-07-19:
I want to build custom rpi AIS Receiver firmware, but i live far from the sea.

sandman7920, 2017-07-19:
I have available
SR161,Comar AIS Receiver,rtl_sdr
and some samples in wav format.
I will be happy to help with whatever I can.

P.S.
My RF knowledge is limited.

F5OEO, 2017-07-19:
My AIS knowledge is limited for now...but as soon as GMSK is implemented, get in touch !

gregology, 2017-11-23:
Following... I'm also interested in AIS tx ⛵🤠

F5OEO, 2017-11-23:
This seems to be possible to implement AIS tx. There is a gnuradio component to do that. For proof of concept, you could use Gnuradio -> IQ -> Rpitx.

pyrog, 2019-01-10:
AIS is also used on land 😉, e.g. [Live map](https://www.marinetraffic.com/en/ais/home/centerx:7.567/centery:47.587/zoom:13) on Rhein river.
See [River Information Services (RIS)](https://www.ccr-zkr.org/13020700-en.html).

liquid-dip have a [GMSK modem](http://liquidsdr.org/doc/gmskmodem/) but how to interface it with rpitx ?

unxs0, 2021-06-10:
Any news on this thread?