F5OEO/rpitx #75: Modulate pll instead clock divider leads to a pure signal

samirbonnet, 2017-07-19:
Is it technically possible to implement this commit from a fork of PiFmRds to Rpitx ?
[https://github.com/SaucySoliton/PiFmRds/commit/65cc5be41da0c97a9a384c5b95a2b442b8e4c568](url)

The result of pll modulation is just amazing. Only harmonics emissions, no more spurious, intermod, phase noise...
Bonus, carrier can go up to 650MHz, rpitx is limited to 250 before switching to harmonic mode.
With it, the RF output looks almost like a regular transmitter and is no more scary. 

Thank you for your amazing work.

F5OEO, 2017-07-20:
Thx very much for pointing this commit : I used this by device tree but that's only for one time. I was inspecting the source code from driver but didn't make this amazing work to port it to userland. 
I will try to include it on rpitx...just wonder if PLL is enough fast to modulate at high bandwidth.

samirbonnet, 2017-07-20:
Thanks, for people who didnt follow the story from the beginning. (pifm then dma then iq mixing emulation)
Creating a development guide, high lvl design or a block diagram will allow you to outsource this kind of work, the creation of a rpitx dev team or even for education purposes !

In IQ Transmitters RF output is achieved by summing two carrier I*cos(wc) and Q*sin(wc) which means hidden in rpitx code there should be 3 amplitude controllers.

1 for summing I*cos(wc) and Q*sin(wc)        result (I*cos(wc)+Q*sin(wc)) >>> direct-RF yay
1 to control amplitude of cos carrier              result I*cos(wc)
1 to control amplitude of sin carrier               result Q*sin(wc)

And obviously rpitx works, as I got no issues Txing some strange/beautiful continuous-phase PSK modulations with it... can you tell us how do you manage to get such amazing result with only frequency control.

samirbonnet, 2017-07-20:
Of course if making pdf is not your thing, I can do the work for you, good documentation will ever be better than exhaustive comments in source code. Creating an IQ mixer with just a square generator as a lot of applications. If you want feel free to contact me in french.

F5OEO, 2017-07-20:
You can contact via email.

F5OEO, 2018-11-23:
Thx again as this is the beginning of the librpitx and V2 version

closed 2018-11-23 by F5OEO