F5OEO/rpitx #76: install.sh

artemipanchuk, 2017-07-22:
WTF? Why there is `apt-get` in install.sh? This is totally incorrect. I am using Arch Linux ARM, for example. There is no `apt`. =)

ha7ilm, 2017-07-22:
You can also build and install this project by `make` and `sudo make install`, as in `install.sh`. I don't think it's a problem that `install.sh` makes it easier for Debian/Ubuntu users.

ha7ilm, 2017-07-22:
It was also possible to use a wrapper like [pacapt](https://github.com/icy/pacapt) (although I haven't tried that yet for my own projects).

artemipanchuk, 2017-07-22:
Yes, of course, there's no problem to edit and build it other way. I was just surprised to see such stuff in install script. )

F5OEO, 2017-07-22:
sorry for this WTF..just a newbie in linux. Let's contribute to become universal...(no raspbian only) !

closed 2018-11-23 by F5OEO