F5OEO/rpitx #77: Transmit DAB (Digital Audio Broadcasting)?

andimik, 2017-08-13:
See closed issue at https://github.com/F5OEO/rpidatv/issues/38

Would it be possible to send a raw file (for example from https://sdr.kt.agh.edu.pl/sdrdab-decoder/download.html ) to GPIO in order to get a Digital Audio Broadcasting stream at a given frequency (for instance 229,072 MHz)?

BW should be 1,7 MHz, Samples always are 2048 kHz.

F5OEO, 2017-08-14:
Interesting idea. Have to test : 2 main potential issues : sample rate at 2MSymbols is not reliable with the current version ; Poor dynamic amplitude could disturb OFDM modulation.

andimik, 2017-11-05:
Is this still utopic?

F5OEO, 2017-11-05:
Need to make a special version to test 2048KHZ I/Q sending. But surely could be an interesting project. So it could maybe possible.

SopaXorzTaker, 2017-11-24:
2048 kHz I/Q would also make it possible to modulate some low-resolution analog TV.

EA1GHE, 2018-03-02:
Any chance to have this special version 2048KHz I/Q?

JsBergbau, 2018-04-13:
I'm also interested.
When you capture an IQ file from an existing DAB broadcast and replay, does this work? Has anyone already tried?

andimik, 2018-04-17:
Tried last year (before I started this issue) but it doesn't work.

andimik, 2025-03-08:
Any idea?

KubaPro010, 2025-03-12:
maybe we can use https://github.com/Opendigitalradio/ODR-DabMod with sendiq? did anyone try that