F5OEO/rpitx #78: Audio Cut-Off

dogebit, 2017-08-23:
I'm using RPITX on my Raspberry Pi 3 with just a single wire, but every-time I try to broadcast it cuts off after a short duration. Usually it takes it about 3 seconds of it playing for it to just cut off completely, even when the audio is longer than that amount of time.

Has anybody else had this problem? 
Thanks.

(Raspberry Pi 3B, Jessie)

matteng1, 2017-08-29:
Same here (RPI3, jessie, single wire).
I have a modified config.txt (force_turbo=1), tried both recommended GPIOs (18, 4) and even installed stretch minimal (horrendous experience with regards to rpitx. When rpitx drops audio and the frequency shifts raspbian stretch gets extremely unresponsive) to no avail.

There is a slight frequency shift just as the audio cuts off.

QuantumRevenger, 2017-09-04:
I'm experiencing the same issue on Pi Zero W with various cable length and both pins documented.

It seems to be generating the signal but stops producing sound around the 6-7 seconds mark for me.

Also sound crackles and pops while playing but goes silent when it stops, ctrl+c to abort returns to normal static noise. Currently a giant RF blocker :/

My best workaround is splitting into 3 second clips and looping through a Python script. This will keep it going but not ideal.

Suggestions most appreciated?

F5OEO, 2017-09-06:
On which Raspbian image do you work ? uname -a

matteng1, 2017-09-07:
(Pi 3 B - v1.2)
Linux raspi3 4.9.45-v7+ #1031 SMP Fri Aug 25 19:02:16 BST 2017 armv7l GNU/Linux

QuantumRevenger, 2017-09-09:
I'm away until Monday and will update/edit with the exact Raspbian Images on my Pi's soon.

In the meantime, if you have any debugging/log commands for me to note, I would be happy to run those accross several images / pi-boards to compare.

dogebit, 2017-09-09:
Sorry, I have been away for a while and unable to be online that much.

I will try to get back with you and run those commands stated.
Thanks.

Nippius, 2017-09-10:
Hi!

I have the same problem. I'm using a RPi3 but I'm running arch linux.

uname -a:
> Linux harbinger 4.9.43-1-ARCH #1 SMP Fri Aug 18 01:10:29 UTC 2017 armv7l GNU/Linux

Not sure if this helps but here is the waterfall when its transmiting the audio as expected:
[image]
And after 6/7 seconds the audio stops and there's only the carrier wave.
[image]

I used the included testfm.sh for generating the signal and the images belong to the output of that script but it's the same if I use testam.sh.

F5OEO, 2017-09-10:
@Nippius thx for spectrum. Didn't have to check but my main idea is with new kernel (4.9) or device tree DMA used to mudulate could be used by an other process. More info in one week or 2.

dogebit, 2017-09-10:
`name -a` gives
`Linux raspberry 4.9.24-v7+ #993 SMP Wed Apr 26 18:01:23 BST 2017 armv7l GNU/Linux`

I'm just using a plain wire, if that helps.

F5OEO, 2017-09-22:
Seems that DMA used in Debian Stretch has changed compare to Jessie. Workaround (to have your feedback) is to add to your rpitx command line : -a 14 (to use DMA 14 instead of 5 by default).

Nippius, 2017-10-15:
Sorry for the delay.

I can confirm that addind -a 14 works perfectly. I've left it run for an hour and the loop never stops :)
I tested this only on the Arch distro because I don't have a debian install available.

Patrusken, 2017-11-11:
Hey! I've followed all the tips above without success. The comand -a 14 helped a little. But have not solved the problem. I feel that when I broadcast the playback speed varies. The beat of the music is getting faster and slower. If I not use the -a 14 command, it plays only a few seconds at a fast varies pace before the "gpio freaks out?" and the result becomes a dead carrier some where randome in the bandwith that disappears when the broadcast is complete. I only get sound output via FM. I have tested SSB (which just sounds like noise). The SSTV sounds uneven too and does not work.

Nippius, 2017-11-14:
Are you doing anything else at the same time? like browsing the web or something similar? The RPi isn't a powerfull system when compared to a normal computer and if it needs to do many things, it will bog down :(

Try closing as many windows and processes as possible and see if it helps.

F5OEO, 2017-11-14:
Thanks @Nippius for feedback. Effectively my web browsing is intensive helping me on many other projects. Have a good FIFO task on the desktop. But contributing could also speed up the development of rpitx, you are welcome. Sure, I close the Windows last development soon https://twitter.com/F5OEOEvariste/status/930460131950845952

Patrusken, 2017-11-14:
That´s not the problem.. I was runing Raspian desktop.. vnc.. and stuff... now i tryd to boot it up clean consol with keboard and screen.. Same thing.. :S

Patrusken, 2017-11-14:
Ok!! I had to do the ./pifm agen with a clean boot up! Now it works! Thanks!

closed 2018-11-23 by F5OEO