F5OEO/rpitx #82: RPitx modulation very poor

PA2LB, 2017-09-28:
I am trying to get rpitx running properly.
When running it in VFO mode it will perform pretty good.
But when trying to transmit a wav file the modulation is very poor , distorted.
The wav i use has a 48k sample rate 
I run rpitx like this: sudo rpitx -i file.wav -m IQ -f 144600 -s 48000 -l 
or like this :  sudo rpitx -i file.wav -m IQ -f 144600  -l -c 1
or like this : sudo rpitx -i file.wav -m IQFLOAT -f 144600 -l
or like this: sudo rpitx -i file.ft -m IQ -f 144600 -s 48000 -l 
or different combinations of the above.

All with the same result ; very poor modulation just squeeqs and distortion.

Any ideas how to fix this?

My final plan is to turn an few rpi's into 2M ARDF beacons 
with filtering etc of course.

73 
PA2LB
Lute van de Bult

F5OEO, 2017-09-28:
Rpitx doesnt accept wav directly. You need to modulate it before : if you want a FM modulator : ./pifm file.wav fm.ft
and then sudo ./rpitx -m RF -i fm.ft -f 144600 -l -c 1
(-c 1 is to ouput to GPIO 4 (Pin 7 oh header))

tony1tf, 2017-10-24:
[SDRSharp_20171024_153327Z_70200550Hz_AF.wav.zip](https://github.com/F5OEO/rpitx/files/1411493/SDRSharp_20171024_153327Z_70200550Hz_AF.wav.zip)
Congrats on a great piece of work, which I have only just discovered. I am running this on an RPi2.
I did the correct conversion of the 48ksps .wav file to the IQ one and the SSB tx worked OK but quite wide bandwidth with distortion, particularly at low frequencies, so I am not sure that any internal RPi filters are working properly. Subsequently, I used Audacity to bandpass my input file from 300 - 3300 Hz with 12dB/octave roll-off. I also used a X5 compression to get better communication quality. Now the result sounds reasonable, but still quite distorted - I have attached an audio file, received with an AirSpy receiver. I have 2 further issues - the -p command doesn't seem to do anything to correct any RPi xtal offset, and the SSB modulation is approx 350Hz higher than the VFO o/p at 70 MHz - ie I have to tune my rx 350Hz up from the carrier.
Tony G3OVH

F5OEO, 2017-10-24:
Can you first try the dev branch :
git checkout dev

Output is by default on GPIO4 (Pin7) instead of GPIO18

Is the example sounds like the same ? TestSSB.sh

tony1tf, 2017-10-25:
[SDRSharp_20171025_212922Z_70200450Hz_AF.zip](https://github.com/F5OEO/rpitx/files/1416264/SDRSharp_20171025_212922Z_70200450Hz_AF.zip)
Hi Evariste
 On the master branch I found what was wrong before - I had been picking up the signal by putting a probe near the GPIOs. However if I connect a wire to the actual GPIO pin, then the RF is not distorted. Previously my AirSpy must have been picking up lots of noise from other pins and the chip itself. Also now the SSB tuning is the same as for the VFO. There is still quite a of of signal several kHz from the main signal, though.  The only remaining problem is to get the -p offset working. I have attached an audio sample again. 
I tried the dev branch, and the code seemed to work, in as much as diagnostic text appeared in the terminal window but no RF output - SSB or VFO, on either of the designated GPIO pins.
Tony

tony1tf, 2017-10-26:
More to my comment on the dev branch - I used SpectrumSpy to find the transmission, if any. It turned out that while it should have been transmitting on 70.200 MHz, it was actually on 63.818 MHz. _ tried -m VFO but that does not give any signal. I adjusted the frequency to transmit on 70.000 MHz, and needed to set -f 77000.00 - exactly 7MHz higher than the output. The resultant transmission was considerably more distorted than the good one using the master branch.

closed 2018-11-23 by F5OEO