F5OEO/rpitx #87: Wrong frequency in dev branch

tony1tf, 2017-10-27:
I raised this in the discussion on distorted signals, but I though it was worth cross-posting here to bring it to the attention of the developer(s).
Using the dev branch on a RPi 2 - I used SpectrumSpy to find the transmission, if any. It turned out that while it should have been transmitting on 70.200 MHz, it was actually on 63.818 MHz. _ tried -m VFO but that does not give any signal. I adjusted the frequency to transmit on 70.000 MHz, and needed to set -f 77000.00 - exactly 7MHz higher than the output.
I did some further tests, setting particular frequencies, and roughly measuring the frequency out with SpectrumSpy:
50.0 MHZ gives 45.49 MHz
60.0 -> 54.564
70.0 -> 63.64
80.0 -> 72.749
90.0 -> 81.827
120.0 -> 109.092
140.0 -> 127.277
These equate to a factor of 0.9091, or 1/1.1. I have started looking at the source code, but it is a bit complex for me to see where the problem lies. BTW, is there a block diagram better showing the main functional parts of the code?

F5OEO, 2017-10-27:
have you made a sudo make install ? Seems that dtblob.bin is not present on /boot folder (setting the master PLL frequency)

tony1tf, 2017-10-27:
Thanks for the advice - I had done various makes and installs, but possible not in the correct order. I finally got the following, and realised I hadn't rebooted the Pi. It now works perfectly - thank you for your help.

!!! You should reboot if it is the first installation !!!
install -m 0755 ../pisstv /usr/bin
install -m 0755 ../pifm /usr/bin
install -m 0755 ../piam /usr/bin
install -m 0755 ../pissb /usr/bin
install -m 0755 ../pifsq /usr/bin
install -m 0755 ../rpitx /usr/bin
install -m 0755 ../pidcf77 /usr/bin
install -m 0755 ../piopera /usr/bin
cp dt-blob.bin /boot/

Tony

tony1tf, 2017-10-27:
And the -p frequency offset is working OK - My Pi seems to need around -5.0
Thanks again.
Tony

closed 2017-11-04 by tony1tf