F5OEO/rpitx #88: only a question about the GPIO pin(s)

beta-tester, 2017-11-04:
i am very new in RPI radio transmission.
and i am wondering, why those projects are using different GPIO pins?
GPIO4 (header pin 7)in rpitx (PWM mode),
GPIO18 (header pin 12) in rpitx (CLK mode),
GPIO12 (header pin 32) in rpidatv,

is there a rason, why there are different GPIOs used all the time?

virendera, 2018-11-15:
Someone confirm me that rpitx works on banana pro or not?

F5OEO, 2018-11-15:
NO : only on Raspberry pi : https://github.com/F5OEO/rpitx/issues/39#issuecomment-247683641

beta-tester, 2018-11-24:
i don't know, how it does answering my question i aksed initially ???

F5OEO, 2018-11-24:
There are main 2 modes : Clock and PWM.
Clock is used in mostly all modulation of rpitx : GPIO is GPIO4
Phased modulation (QPSK for datv for example) is easy to do with PWM. It is used in rpidatv and rpitx in datv mode. GPIO 12 or GPIO18 could output this PWM signal

beta-tester, 2018-11-24:
thank you for explaining me the background.