F5OEO/rpitx #89: usage in cubesats

rnbokade, 2017-11-13:
can this be used with some good filter and an amplifier circuit to use with lower earth orbit cubesat

cipiricus, 2018-03-06:
If you want to use Raspberry Pi as main board in a cubesat will be useless, because if i'm not wrong all satellites use a special radiation hardened electronics which are very expensive, Raspberry Pi is not designed for that purpose

ha7ilm, 2018-03-07:
Power efficiency is also questionable if using it on-board.

I guess @rnbokade would rather use it from the ground to communicate with a cubesat, right?

lu9cbl, 2018-03-20:
You can use the raspberry py CM3 (Computer Module 3), that it's designed to special enviroments. Maybe it's works in a Cubesat.

closed 2021-03-08 by rnbokade