F5OEO/rpitx #92: Rpitx signal getting lower and lower

bratlemi, 2018-01-03:
Hi,
Used rpitx to output a song converted to wav and play on my stereo. FM mode, chose 88Mhz (no station there)
Worked better than expected. Without wire,  in a case, across the room - decent signal.
Next day, i had to put a wire to GPIO18 because it gave too weak signal.
Day after, signal from gpio18+wire is so weak that i have to connect my stereo antenna with the wire... no transmit, basicaly a transfer :(
Please help, i really love this tool.
I tried pifmrds but that gives me no output whatsoever

klewisjohnson, 2018-01-03:
It worked fine the first time.  Through air?

bratlemi, 2018-01-03:
Yes. Worked great. Signal good, across the room. Next day much lower, day after that it transmits to like 1cm away with a lot of static. Really really weak output signal.
Same input file, same command

bratlemi, 2018-01-07:
Help please