F5OEO/rpitx #95: pifm modulation makes audio too quiet

Nawor3565, 2018-01-13:
So, I'm trying to make a short-range FM audio transmitter to play music in my car. I've gotten everything set up and working, and it can broadcast a pretty powerful FM signal. I'm using an RTL-SDR on my PC to test it out, and there's one problem. Despite the signal being strong, the volume of audio is really low compared to normal FM stations. I can't find much information on Google about it, but I'm pretty sure it's a problem with the modulation. Comparing the signal from the Pi (left) and a local radio channel (right), you can see how the bandwidth of the Pi's signal is much smaller.

[image]

Mind, I can still hear it fine, I just have to turn the volume way up. But that's a problem when I then change to a different channel and blow my ears out. Just another example, I took audio clips from both stations at my SDR's "normal" volume level, and while I can hear the local channel fine, the Pi's is barely audible. 

[image]

I've tried it with a few different music files, and editing them with Audacity to increase the volume, but it doesn't seem to make a difference. Do you have any suggestions on how I could fix this? Thanks!

F5OEO, 2018-01-14:
Thx for analysis ... I know this issue.
dev version should be better : git checkout dev && git pull
default GPIO pin is now GPIO4 (Pin Header 7).
The main problem is demphasis and FM excursion. I mainly implement NFM for an amateur radio usage, i should add a WFM to permit a wider excursion : see in https://github.com/F5OEO/rpitx/blob/b54300c9e6baecfd6ee4b886e152536567740a90/fm/pifm.c#L88
Increasing this value should solve the issue for WFM

sm3ulc, 2018-01-14:
Hi!

For that purpose I recommend pifmadv. It's complete with
wfm(mono/stereo)+rds.

https://github.com/Miegl/PiFmAdv

73

// David

Nawor3565, 2018-01-15:
F5OEO - Just tried the dev version, it seems to work better, but like someone else said the frequency it actually transmits at is different from what I set. Unfortunatly I still couldn't quite get WFM working, but I could have been doing something wrong. Do I need to edit the Excursion value and then run install.sh again? 

sm3ulc - Thanks! I just tried it out, it works great so I think I'll use it for WFM until I can get this one working

BigNerd95, 2018-01-17:
You can try my implementation of WAV FM transmitter.
https://github.com/BigNerd95/RPiRadio/blob/master/Examples/WAV%20FM/wav_fm.c
I used 75 kHz of deviation from nominal carrier as the standard tells (https://en.wikipedia.org/wiki/Frequency_deviation)

It is also completely commented of you are interested how it works.

F5OEO, 2018-11-23:
FM rewriten in v2

closed 2018-11-23 by F5OEO