F5OEO/rpitx #100: File system gets corrupted

OrbitTheSun, 2018-03-19:
Hello,

i have a Raspberry Pi 3B with stretch, kernal version Linux 4.9.80-v7+, firmware Mar 13 2018 (6e08617e7767b09ef97b3d6cee8b75eba6d7ee0b).

After successfully sending a doorbell command (3 seconds long) for a couple of times, the root file system /dev/mmcblk0p2 was automatically remounted read-only and write protected by the system. Fsck on the read-only FS showed errors. After a reboot with file check and recovery, the system was ok and running.

/var/log/syslog had several suspicious entries with the first entry being:

> mmc0: timeout waiting for hardware interrupt.

I tried to reproduce the error by sending the doorbell sequence again. This time I checked syslog immediately, which showed new error lines:

> Mar 19 10:02:29 raspi2 kernel: [29063.150745] mmc0: timeout waiting for hardware interrupt.
> Mar 19 10:02:29 raspi2 kernel: [29063.150762] [033a6af0] DMA< b945ea40 0
> Mar 19 10:02:29 raspi2 kernel: [29063.150769] [033a6af1] DMA  99 10801
> Mar 19 10:02:29 raspi2 kernel: [29063.150776] [033a6af3] FDA< b945e968 0
> Mar 19 10:02:29 raspi2 kernel: [29063.150783] [033a6af4] TCM< b945ea40 0
> Mar 19 10:02:29 raspi2 kernel: [29063.150789] [033a6af5] CMD< c 0
> Mar 19 10:02:29 raspi2 kernel: [29063.150796] [033a6af7] TCM> b945ea40 0
[...]
> Mar 19 10:02:29 raspi2 kernel: [29063.152471] [0480b8e1] TIM< 0 0
> Mar 19 10:02:29 raspi2 kernel: [29063.152484] mmc0: sbc op 23 arg 0x30 flags 0x15 - resp 00000000 00000000 00000000 00000000, err 0
> Mar 19 10:02:29 raspi2 kernel: [29063.152494] mmc0: cmd op 25 arg 0x4748b8 flags 0xb5 - resp 00000900 00000000 00000000 00000000, err 0
> Mar 19 10:02:29 raspi2 kernel: [29063.152500] mmc0: data blocks 30 blksz 200 - err 0
> Mar 19 10:02:29 raspi2 kernel: [29063.152510] mmc0: stop op 12 arg 0x0 flags 0x49d - resp 00000000 00000000 00000000 00000000, err 0
> Mar 19 10:02:29 raspi2 kernel: [29063.152514] mmc0: =========== REGISTER DUMP ===========
> Mar 19 10:02:29 raspi2 kernel: [29063.152519] mmc0: SDCMD  0x00000099
> Mar 19 10:02:29 raspi2 kernel: [29063.152523] mmc0: SDARG  0x004748b8
> Mar 19 10:02:29 raspi2 kernel: [29063.152528] mmc0: SDTOUT 0x017d7840
> Mar 19 10:02:29 raspi2 kernel: [29063.152532] mmc0: SDCDIV 0x00000003
> Mar 19 10:02:29 raspi2 kernel: [29063.152537] mmc0: SDRSP0 0x00000900
> Mar 19 10:02:29 raspi2 kernel: [29063.152541] mmc0: SDRSP1 0x00001918
> Mar 19 10:02:29 raspi2 kernel: [29063.152546] mmc0: SDRSP2 0xffffffff
> Mar 19 10:02:29 raspi2 kernel: [29063.152551] mmc0: SDRSP3 0x0002400f
> Mar 19 10:02:29 raspi2 kernel: [29063.152555] mmc0: SDHSTS 0x00000001
> Mar 19 10:02:29 raspi2 kernel: [29063.152560] mmc0: SDVDD  0x00000001
> Mar 19 10:02:29 raspi2 kernel: [29063.152564] mmc0: SDEDM  0x0001080a
> Mar 19 10:02:29 raspi2 kernel: [29063.152569] mmc0: SDHCFG 0x0000040e
> Mar 19 10:02:29 raspi2 kernel: [29063.152573] mmc0: SDHBCT 0x00000200
> Mar 19 10:02:29 raspi2 kernel: [29063.152578] mmc0: SDHBLC 0x00000030
> Mar 19 10:02:29 raspi2 kernel: [29063.152582] mmc0: ===========================================
> Mar 19 10:02:29 raspi2 kernel: [29063.152887] mmcblk0: error -110 transferring data, sector 4671672, nr 48, cmd response 0x900, card status 0xc00

After reboot and fsck I tried again and succeded to reproduce the mmc0 lines in syslog, kern.log and messages.

Has anyone also observed this behavior? - I was using the current version of rpitx from master.

-Andreas

F5OEO, 2018-03-19:
On latest kernel be sure to use -a 14 , else rpitx use DMA shared with SD card. rpitx is now completely rewriting using librpitx. Will be update very soon.

OrbitTheSun, 2018-03-19:
Thank you for the quick response. No more error lines in syslog! - However the signal seems weaker than before. I'll check the frequency.

-Andreas

F5OEO, 2018-03-19:
Prefere to use -c 1 , means GPIO 4 for a better signal.

OrbitTheSun, 2018-03-19:
Yes `-c 1` works better, but the bandwidth of the signal is very wide now. Center frequency is 433.935 MHz and the spectrum spreads out from 433.2 to 434.6 MHz. Thus the range is only 50 cm for the doorbell to ring.

Yesterday without `-a 14`, range was around 2 meters. And even without a wire pluged in, range was 50 cm then. As I remember the bandwidth was normal as observed with Gqrx and a rtl-sdr.com dongle on my Linux desktop pc.

Today I do not want to try without `-a 14` any more... ;-).

Branch v2beta does not  currently compile on my RPi 3. Will it be possible to run rpitx without `-a 14` ?

F5OEO, 2018-03-19:
Without -a 14, you will have issue with DMA.
with -c 1 , you need to plug a wire on GPIO 4/Pin 7.
v2rpitx is becoming librpitx, so this branch is obsolete.
I have to rebuild rpitx based on librpitx. Coming soon ;

OrbitTheSun, 2018-03-19:
> you need to plug a wire on GPIO 4/Pin 7.

I tested `-c 1` with a wire pluged into Pin 7.

> I have to rebuild rpitx based on librpitx. Coming soon

Very much appreciated. - Thank you!

F5OEO, 2018-11-23:
V1 related .

closed 2018-11-23 by F5OEO