F5OEO/rpitx #99: System Audio crashes.

Commander-Crash, 2018-03-19:
I notice when using RPITX to transmit that after its done my system audio crashes no sound nothing. On top of that my buttons connected to the GPIO pins no longer respond. Is there are work around for this issue. Im using a RPI3 B. Kernel version: Linux 4.4.34-v7+ armv7 l::: Firmware: #930 :::: Distribution: Raspbian GNU/Linux 8 (jessie)

F5OEO, 2018-03-19:
On latest kernel be sure to use -a 14 , else rpitx use DMA shared with SD card. rpitx is now completely rewriting using librpitx. Will be update very soon.

Commander-Crash, 2018-03-21:
Thanks for your response. -a 14 will do the same with the audio. But I will update my kernel to the latest and try again.

closed 2018-04-06 by Commander-Crash