F5OEO/rpitx #102: rpitx does not transmit anything

Hexalyse, 2018-04-08:
I've been playing `pifm` and `PiFmRds` a lot on my Raspberry Pi and everything works fine.

Today I tried using `rpitx`, but it does not transmit anything. Here is an example output from `testssb.sh`:

```
Reading file : ./sampleaudio.wav
Sample Rate : 48000
Channels    : 1

Writing file : ./ssbIQ.wav
Channels    : 2
rpitx Version 0.2 compiled Apr  4 2018 (F5OEO Evariste) running on  Frequency=0.000000  SampleRate=48000 Master PLL = 1000000000

 Warning : Using harmonic 3
memory: 1024 MB
processor: Broadcom BCM2837
i2cDevice: /dev/i2c-1
model: Model B Pi 3
manufacturer: Embest
pcb revision: 2
warranty void: no
revision: a22082
peripheral base: 0x3f000000
Jessie
Using mbox device /dev/vcio.
3616000 Size NUM PAGES 883 PAGE_SIZE 4096
MASH 1 Freq PLL# 6
Calibrate : ppm=-6 DMA 166ns:1277ns
WaitNano=20833 F1=96426385.423043 TuneFrequency 96428222.222222 F2=96428655.507687 Initial Resolution(Hz)=2270.084644 ResolutionPWMF 18.160677 NbStep=125 DELAYStep=7
****** STARTING TRANSMIT ********
```

I have a RTL-SDR dongle plugged in my desktop computer, and I get no signal on 933.9MHz.

Xeddius, 2018-04-12:
I can also confirm that the current version of rpitx is not transmitting. Directly installed from git.
Using RTL-SDR and a 4 inch antenna from gpio 4 of the raspberry pi.

F5OEO, 2018-04-12:
> Do we have an idea of what version worked ? Maybe we could try checking
> out an older commit, until we find one that works.
> Or maybe it is because of an update in Raspbian ?

A new version (branch v2beta) will be released soon.
In the meantime you can try with master version adding  -c 1 -a 14 , output
RF is on GPIO4 pin 7

Hexalyse, 2018-04-12:
Nice ! Thanks. Yes I tried checking out the v2beta branch and compiling it, but no transmission either (or maybe I messed up).

thailemon, 2018-06-21:
case 'a': // DMA Channel 1-14

acujl, 2018-09-26:
Not working to me either, is the problem fixed?

F5OEO, 2018-11-23:
Fixed with v2

closed 2018-11-23 by F5OEO

Murphyv310, 2019-10-10:
No results at all on a Pi 4 2gb